Azure/Industrial-IoT #973: Microservices fail to recover after a manual/automatic IoT Hub failover

wkoschen, 2020-12-10:
**Describe the bug**
Was testing the high-availability failover feature on the IoT hub.  Microservices were failing after the failover completed and did not recover.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the IoT hub in Azure portal and initiate a manual failover
2. Wait for failover to complete
3. Observe state of microservices

**Expected behavior**
Since the failover is a built in high-availability feature which can be triggered automatically and without notice it is expected that the Azure IIoT microservices will be able to recover from a failover.

**Additional context**
The helm chart was configured to load configuration from KeyVault.  It appears that the Event Hub-compatible endpoint is stored in the configuration.  IoT hub documentation states this value will change after a failover.

Didn't capture all of the logging details as we needed to fix the environment to move on to additional testing.  Can capture specifics as requested.

Here is an example of the error messages we were seeing:

[14:37:34 ERR Microsoft.Azure.IIoT.Hub.Processor.EventHub.EventProcessorHost] Error starting event processor host. 
Microsoft.Azure.EventHubs.Processor.EventProcessorConfigurationException: Encountered error while fetching the list of EventHub PartitionIds
 ---> System.Net.Sockets.SocketException (0xFFFDFFFF): Name or service not known
   at Microsoft.Azure.EventHubs.Amqp.AmqpEventHubClient.CreateConnectionAsync(TimeSpan timeout)

wkoschen, 2020-12-11:
We're trying to repair that environment now.  We thought updating the endpoint in the keyvault would do the trick but now we see the following.  Tried creating new key and redeploying the chart with explicit values but no luck.  Not sure what is "Unauthorized".

This is from the telemetry-cdm-processor logs.

Unhandled exception. Microsoft.Azure.EventHubs.Processor.EventProcessorConfigurationException: Encountered error while fetching the list of EventHub PartitionIds
 ---> System.UnauthorizedAccessException: Unauthorized
   at Microsoft.Azure.EventHubs.Amqp.Management.AmqpServiceClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.EventHubClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()

karok2m, 2020-12-14:
@wkoschen, unfortunately this functionality is not supported at the moment, as the value of Event Hub-compatible endpoint is passed to the chart. So if the value is changed after failover of IoT Hub, then IIoT platform will not be able to automatically pick it up.

The secret in Key Vault that stores the value for Event Hub-compatible endpoint is called `pcs-iothub-eventhubendpoint`. If you update its value and restart all microservices, then the new value should be picked up. But if you change the chart so that all required configuration parameters are passed directly to the chart, then please pay attention to the value of `loadConfFromKeyVault` parameter. As it will either enable or disable loading of configuration from Key Vault. More details about `loadConfFromKeyVault` can be found [here](https://github.com/Azure/Industrial-IoT/blob/master/deploy/helm/azure-industrial-iot/README.md#load-configuration-from-azure-key-vault).

marcschier, 2023-01-16:
We are not planning to support IoT Hub failover handling.

closed 2023-01-16 by marcschier