Azure/Industrial-IoT #980: "Twin not authorized using given secret" error for OPC Twin module when connecting to kepware

yang-cy-94, 2020-12-15:
**Describe the bug**
having issue when connecting OPC Twin to kepware. 
an OPC Publisher module and OPC Twin module are deployed in and edge device (edge device is deployed as a transparent gateway with OS Ubuntu 18.04), OPC Publisher is able to connect to the kepware but OPC Twin is having authentication issue. edge device root certificate is added into the kepware trusted root store. reference link: https://docs.microsoft.com/en-us/azure/iot-accelerators/howto-opc-twin-deploy-existing

Please check following the authentication error for the OPC Twin module:
[05:30:16 INF Discovery] Twin not authorized using given secret - exit host...
	Microsoft.Azure.Devices.Client.Exceptions.UnauthorizedException: CONNECT failed: RefusedNotAuthorized
	at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsyncInternal(CancellationToken cancellationToken)
	at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsync(CancellationToken cancellationToken)
	at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
	at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass23_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
	--- End of stack trace from previous location where exception was thrown ---
	at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
	at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass33_0.<<OpenInternalAsync>b__0>d.MoveNext()
	--- End of stack trace from previous location where exception was thrown ---
	at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.EnsureOpenedAsync(CancellationToken cancellationToken)
	at Microsoft.Azure.Devices.Client.InternalClient.OpenAsync()
	at Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory.DeviceClientAdapter.CreateAsync(String product, IotHubConnectionStringBuilder cs, String deviceId, ITransportSettings transportSetting, TimeSpan timeout, IRetryPolicy retry, Action onConnectionLost, ILogger logger) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Default\IoTSdkFactory.cs:line 476
	at Microsoft.Azure.IIoT.Utils.Try.Options[T](Func`1[] options)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Please find the trusted root certificate for the transparent gateway added to the local computer where the Kepware server is running.
 
[image]

Below is the screenshot of the MQTT agent connecting to the transparent edge gateway.
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

yang-cy-94, 2020-12-15:
please check support_bundle:
[support_bundle (3).zip](https://github.com/Azure/Industrial-IoT/files/5693751/support_bundle.3.zip)

marcschier, 2021-02-26:
Hi - please use version 2.7.206 for the services and OPC Twin/Publisher images.   The device scope issue that causes this behavior was fixed there as well as in master.

closed 2021-02-26 by marcschier