Azure/Industrial-IoT #988: Unable to Authroze Swagger UI

nadeem4, 2020-12-18:
**Describe the bug**
I am unable to Authorize Swagger. It is saying that  ""error":"invalid_resource","error_description":"AADSTS500011:+The+resource+principal+named+https://xxxxxxxxxxxxxxx/<azure_ad_app_name>/+was+not+found+in+the+tenant+named+<tenant>.+This+can+happen+if+the+application+has+not+been+installed+by+the+administrator+of+the+tenant+or+consented+to+by+any+user+in+the+tenant.+You+might+have+sent+your+authentication+request+to+the+wrong+tenant.".

I deployed the same app, 2 months back, and it worked completely fine.

 I found one link with a similar problem. They had extra '/' at the end, and that was the issue, and I checked the resource URL, it has extra '/' at the end.

https://community.powerbi.com/t5/Developer/AADSTS500011-The-resource-principal-named-http-analysis-windows/m-p/1347727#M25552

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to 'swagger UI of IIOT service.'
2. Click on 'Authorize.'
3. Scroll down to 'Check out the error.'
4. See an error

**Expected behaviour**
A clear and concise description of what you expected to happen.

I should be able to Authorize.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Windows
 - Browser [e.g. chrome, safari] Chrome
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6] NA
 - OS: [e.g. iOS8.1] NA
 - Browser [e.g. stock browser, safari] NA
 - Version [e.g. 22] NA

**Additional context**
Add any other context about the problem here.

nadeem4, 2020-12-23:
Does anybody face a similar issue?

marcschier, 2023-05-19:
Please retry with 2.9.0-preview2 release, we switched to use MSAL and MSGraph for registration.

closed 2023-05-19 by marcschier