Azure/Industrial-IoT #992: Telemetry message count is not in sync for every Batch having same Scan Rate - OPCUA

PayalDodeja, 2020-12-22:
**Describe the bug**
Different batches with same scan rate have different Message Count

**To Reproduce**
Steps to reproduce the behavior:
1. Publish the node with with following publishing interval and sampling interval **[ Scan Rate]** 

> {
>             "nodeId": "nsu=KEPServerEX;s=ATChannel.ATDevice.RampDouble",
>             "publishingInterval": "00:00:01",
>             "samplingInterval": "00:00:00.5000000"
>     }

2. Now telemetry messages should start flowing in and every batch should contain just 1 message

> **[IoTHubMonitor]** [5:22:36 PM] Message received from [DataSetWriterV2_uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b]:
> [
>   {
>     "NodeId": "nsu=KEPServerEX;s=ATChannel.ATDevice.RampDouble",
>     "EndpointUrl": "opc.tcp://10.1.0.5:49320/",
>     "ApplicationUri": "urn:pegasus-azure-d:Kepware.KEPServerEX.V6:UA%20Server",
>     "DisplayName": "RampDouble",
>     "Timestamp": "2020-12-22T22:22:35.3117603Z",
>     "Status": "BadMonitoredItemIdInvalid",
>     "Value": {
>       "StatusCode": {
>         "Symbol": "BadMonitoredItemIdInvalid",
>         "Code": 2151809024
>       }
>     },
>     "SequenceNumber": 1,
>     "ExtensionFields": {
>       "EndpointId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "PublisherId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "DataSetWriterId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b"
>     }
>   }
> ]

3. If you change the publishing interval and sampling interval **[ Scan Rate]**   for the published node

> {
>     "items": [
>         {
>             "nodeId": "nsu=KEPServerEX;s=ATChannel.ATDevice.RampDouble",
>             "publishingInterval": "00:00:00.1000000",
>             "samplingInterval": "00:00:00.0500000"
>         }
>     ]
> }

4. Number of messages in each block may vary. In the below example, you will see 2 messages in the first block and 5 messages in the next Batch

> **[IoTHubMonitor]** [5:22:26 PM] Message received from [DataSetWriterV2_uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b]:
> [
>   {
>     "**_NodeId_**": "nsu=KEPServerEX;s=ATChannel.ATDevice.RampDouble",
>     "EndpointUrl": "opc.tcp://10.1.0.5:49320/",
>     "ApplicationUri": "urn:pegasus-azure-d:Kepware.KEPServerEX.V6:UA%20Server",
>     "DisplayName": "RampDouble",
>     "Timestamp": "2020-12-22T22:22:25.2487173Z",
>     "Status": "Good",
>     "Value": {
>       "Value": 206.19999999999257,
>       "SourceTimestamp": "2020-12-22T22:22:25.1837061Z",
>       "ServerTimestamp": "2020-12-22T22:22:25.1837061Z"
>     },
>     "SequenceNumber": 1583,
>     "ExtensionFields": {
>       "EndpointId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "PublisherId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "DataSetWriterId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b"
>     }
>   },
> 
> 
> 
>   {
>     "**_NodeId":_** "nsu=KEPServerEX;s=ATChannel.ATDevice.RampDouble",
>     "EndpointUrl": "opc.tcp://10.1.0.5:49320/",
>     "ApplicationUri": "urn:pegasus-azure-d:Kepware.KEPServerEX.V6:UA%20Server",
>     "DisplayName": "RampDouble",
>     "Timestamp": "2020-12-22T22:22:25.3603704Z",
>     "Status": "Good",
>     "Value": {
>       "Value": 206.29999999999256,
>       "SourceTimestamp": "2020-12-22T22:22:25.3040421Z",
>       "ServerTimestamp": "2020-12-22T22:22:25.3040421Z"
>     },
>     "SequenceNumber": 1584,
>     "ExtensionFields": {
>       "EndpointId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "PublisherId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "DataSetWriterId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b"
>     }
>   }
> ]
> 
> 
> 
> **[IoTHubMonitor]** [5:22:26 PM] Message received from [DataSetWriterV2_uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b]:
> [
>   {
>     "**_NodeId_**": "nsu=KEPServerEX;s=ATChannel.ATDevice.RampDouble",
>     "EndpointUrl": "opc.tcp://10.1.0.5:49320/",
>     "ApplicationUri": "urn:pegasus-azure-d:Kepware.KEPServerEX.V6:UA%20Server",
>     "DisplayName": "RampDouble",
>     "Timestamp": "2020-12-22T22:22:26.2293573Z",
>     "Status": "Good",
>     "Value": {
>       "Value": 207.09999999999252,
>       "SourceTimestamp": "2020-12-22T22:22:26.184022Z",
>       "ServerTimestamp": "2020-12-22T22:22:26.184022Z"
>     },
>     "SequenceNumber": 1585,
>     "ExtensionFields": {
>       "EndpointId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "PublisherId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "DataSetWriterId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b"
>     }
>   },
> 
> 
> 
>   {
>     "**_NodeId_**": "nsu=KEPServerEX;s=ATChannel.ATDevice.RampDouble",
>     "EndpointUrl": "opc.tcp://10.1.0.5:49320/",
>     "ApplicationUri": "urn:pegasus-azure-d:Kepware.KEPServerEX.V6:UA%20Server",
>     "DisplayName": "RampDouble",
>     "Timestamp": "2020-12-22T22:22:26.3345816Z",
>     "Status": "Good",
>     "Value": {
>       "Value": 207.1999999999925,
>       "SourceTimestamp": "2020-12-22T22:22:26.3145652Z",
>       "ServerTimestamp": "2020-12-22T22:22:26.3145652Z"
>     },
>     "SequenceNumber": 1586,
>     "ExtensionFields": {
>       "EndpointId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "PublisherId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "DataSetWriterId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b"
>     }
>   },
> 
> 
> 
>   {
>     "**_NodeId_**": "nsu=KEPServerEX;s=ATChannel.ATDevice.RampDouble",
>     "EndpointUrl": "opc.tcp://10.1.0.5:49320/",
>     "ApplicationUri": "urn:pegasus-azure-d:Kepware.KEPServerEX.V6:UA%20Server",
>     "DisplayName": "RampDouble",
>     "Timestamp": "2020-12-22T22:22:26.4547671Z",
>     "Status": "Good",
>     "Value": {
>       "Value": 207.2999999999925,
>       "SourceTimestamp": "2020-12-22T22:22:26.4295737Z",
>       "ServerTimestamp": "2020-12-22T22:22:26.4295737Z"
>     },
>     "SequenceNumber": 1587,
>     "ExtensionFields": {
>       "EndpointId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "PublisherId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "DataSetWriterId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b"
>     }
>   },
> 
> 
> 
>   {
>     "**_NodeId_**": "nsu=KEPServerEX;s=ATChannel.ATDevice.RampDouble",
>     "EndpointUrl": "opc.tcp://10.1.0.5:49320/",
>     "ApplicationUri": "urn:pegasus-azure-d:Kepware.KEPServerEX.V6:UA%20Server",
>     "DisplayName": "RampDouble",
>     "Timestamp": "2020-12-22T22:22:26.5620257Z",
>     "Status": "Good",
>     "Value": {
>       "Value": 207.3999999999925,
>       "SourceTimestamp": "2020-12-22T22:22:26.489862Z",
>       "ServerTimestamp": "2020-12-22T22:22:26.489862Z"
>     },
>     "SequenceNumber": 1588,
>     "ExtensionFields": {
>       "EndpointId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "PublisherId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "DataSetWriterId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b"
>     }
>   },
> 
> 
> 
>   {
>     "**_NodeId_**": "nsu=KEPServerEX;s=ATChannel.ATDevice.RampDouble",
>     "EndpointUrl": "opc.tcp://10.1.0.5:49320/",
>     "ApplicationUri": "urn:pegasus-azure-d:Kepware.KEPServerEX.V6:UA%20Server",
>     "DisplayName": "RampDouble",
>     "Timestamp": "2020-12-22T22:22:26.6708259Z",
>     "Status": "Good",
>     "Value": {
>       "Value": 207.4999999999925,
>       "SourceTimestamp": "2020-12-22T22:22:26.61813Z",
>       "ServerTimestamp": "2020-12-22T22:22:26.61813Z"
>     },
>     "SequenceNumber": 1589,
>     "ExtensionFields": {
>       "EndpointId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "PublisherId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b",
>       "DataSetWriterId": "uat3942fddb971f4cefe4f6de8bbc3077a2ac979b9b"
>     }
>   }
> ]

**Expected behavior**
Number of Telemetry Messages should be same for every Batch

mregen, 2021-01-05:
Hi @PayalDodeja, I need more information about your setup. 
-Which version are you using?
-orchestrated or standalone mode?
-samples or pubsub encoding?

monikavar, 2021-01-19:
Closing due to inactivity. If there is an issue, please re-open.

PayalDodeja, 2021-01-19:
Hello @mregen  , apologies for very late  response to your question above.
To answer them now: 
-Which version are you using? ---->  **this would be 2.7.183 version of publisher module**
-orchestrated or standalone mode? ------> **orchestrated** 
-samples or pubsub encoding? --------> **pubsub as far as I know because we are using default config .... I am not quite aware of samples encoding** 

@monikavar I think I don't see an option to re-open the bug, should I create a new issue and link this one to it ?

marcschier, 2021-03-30:
Hi @PayalDodeja - could you use the latest release 2.7.206 instead and see if the behavior is the same?  There are several fixes in 2.7.206 which relate to publishing configuration, specifically to sampling rate.

PayalDodeja, 2021-03-30:
Hi @marcschier , I checked for the behavior with 2.7.206, it is still the same.  For the publishing interval in the snippet below, I see batch sizes varying between 4 and 5.

> `{
>     "items": [
>         {
>             "nodeId": "nsu=ThingWorx Kepware Server;s=ATChannel.ATDevice.RampDouble",
>             "publishingInterval": "00:00:00.1000000",
>             "samplingInterval": "00:00:00.0500000"
>         }
>     ]
> }`

_____________________________size 5 ____________________________________________________________________

> `[IoTHubMonitor] [1:11:04 PM] Message received from [DataSetWriterV2_uatf1d27a233553e06f00ccc20bf3d5e6df6182b791]:
> [
>   {
>     "NodeId": "nsu=ThingWorx Kepware Server;s=ATChannel.ATDevice.RampDouble",
>     "EndpointUrl": "opc.tcp://10.1.0.5:49320/",
>     "ApplicationUri": "urn:Stratus-PDodeja:PTC.ThingWorxKepwareServer:UA%20Server",
>     "DisplayName": "RampDouble",
>     "Timestamp": "2021-03-30T17:11:03.6786969Z",
>     "Status": "Good",
>     "Value": {
>       "Value": 1070.2000000001221,
>       "SourceTimestamp": "2021-03-30T17:11:03.60156Z",
>       "ServerTimestamp": "2021-03-30T17:11:03.60156Z"
>     },
>     "SequenceNumber": 610,
>     "ExtensionFields": {
>       "EndpointId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791",
>       "PublisherId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791",
>       "DataSetWriterId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791_100"
>     }
>   },
>   {
>     "NodeId": "nsu=ThingWorx Kepware Server;s=ATChannel.ATDevice.RampDouble",
>     "EndpointUrl": "opc.tcp://10.1.0.5:49320/",
>     "ApplicationUri": "urn:Stratus-PDodeja:PTC.ThingWorxKepwareServer:UA%20Server",
>     "DisplayName": "RampDouble",
>     "Timestamp": "2021-03-30T17:11:03.7873482Z",
>     "Status": "Good",
>     "Value": {
>       "Value": 1070.300000000122,
>       "SourceTimestamp": "2021-03-30T17:11:03.7246077Z",
>       "ServerTimestamp": "2021-03-30T17:11:03.7246077Z"
>     },
>     "SequenceNumber": 611,
>     "ExtensionFields": {
>       "EndpointId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791",
>       "PublisherId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791",
>       "DataSetWriterId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791_100"
>     }
>   },
>   {
>     "NodeId": "nsu=ThingWorx Kepware Server;s=ATChannel.ATDevice.RampDouble",
>     "EndpointUrl": "opc.tcp://10.1.0.5:49320/",
>     "ApplicationUri": "urn:Stratus-PDodeja:PTC.ThingWorxKepwareServer:UA%20Server",
>     "DisplayName": "RampDouble",
>     "Timestamp": "2021-03-30T17:11:03.8970361Z",
>     "Status": "Good",
>     "Value": {
>       "Value": 1070.400000000122,
>       "SourceTimestamp": "2021-03-30T17:11:03.8363212Z",
>       "ServerTimestamp": "2021-03-30T17:11:03.8363212Z"
>     },
>     "SequenceNumber": 612,
>     "ExtensionFields": {
>       "EndpointId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791",
>       "PublisherId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791",
[... 99 lines cut ...]
>       "EndpointId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791",
>       "PublisherId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791",
>       "DataSetWriterId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791_100"
>     }
>   },
>   {
>     "NodeId": "nsu=ThingWorx Kepware Server;s=ATChannel.ATDevice.RampDouble",
>     "EndpointUrl": "opc.tcp://10.1.0.5:49320/",
>     "ApplicationUri": "urn:Stratus-PDodeja:PTC.ThingWorxKepwareServer:UA%20Server",
>     "DisplayName": "RampDouble",
>     "Timestamp": "2021-03-30T17:11:05.5256747Z",
>     "Status": "Good",
>     "Value": {
>       "Value": 1071.9000000001206,
>       "SourceTimestamp": "2021-03-30T17:11:05.4437643Z",
>       "ServerTimestamp": "2021-03-30T17:11:05.4437643Z"
>     },
>     "SequenceNumber": 627,
>     "ExtensionFields": {
>       "EndpointId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791",
>       "PublisherId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791",
>       "DataSetWriterId": "uatf1d27a233553e06f00ccc20bf3d5e6df6182b791_100"
>     }
>   }
> ]`

koepalex, 2021-08-19:
@PayalDodeja  if you are using the orchestrated mode, then it is the `Samples` format with the `Full Featured Message` flag. 

How did you configure the batch size `PCS_DEFAULT_PUBLISH_JOB_BATCH_SIZE` and batch interval `PCS_DEFAULT_PUBLISH_JOB_BATCH_INTERVAL`?

In general, having the same publishing interval, means that the monitored items using the same subscription. To be part in the output the value of OPC UA node needs to change, therefor it is **never** guaranteed that nodes with the same interval are also together in the same output message.