Azure/Industrial-IoT #993: microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Could not send worker heartbeat

sagarganachari, 2020-12-23:
**Describe the bug**
A clear and concise description of what the bug is.

**Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Could not send worker heartbeat. 
Microsoft.Azure.IIoT.Exceptions.InvalidConfigurationException: Job orchestrator not configured
   at Microsoft.Azure.IIoT.OpcUa.Api.Publisher.Clients.PublisherOrchestratorClient.SendHeartbeatAsync(HeartbeatModel heartbeat, CancellationToken ct) in D:\a\1\s\api\src\Microsoft.Azure.IIoT.Api\src\Publisher\Clients\PublisherOrchestratorClient.cs:line 84
   at Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker.HeartbeatTimer_ElapsedAsync(Object sender) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Agent.Framework\src\Agent\Default\Worker.cs:line 151**

**To Reproduce**
Steps to reproduce the behavior:
i am getting error iotedge logs OPCPublisher command

**Expected behavior**
data from OPC UA server is not reflecting.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - ubuntu 18.1
 

 -
**Additional context**
Please let us know where exactly problem is with this error

sagarganachari, 2020-12-24:
is some one face same issue?? please let me know

marcschier, 2020-12-28:
Publisher module needs the url for the orchestration services to work, the url is passed to the module via its twin from the service.   If the service is not running (e.g. not started, or stopped) then it will not yet have been set.   Ensure the service is running and wait until the twin setting was written and synchronized with the module.   This can take a couple minutes initially.   If the service is not running, ensure you start it.   If you intend to run the publisher module standalone (without a service endpoint, but with configuration from publishednodes.json file, ensure you passed the file name via --pf on the module container command line and the file exists (is accessible via a volume).   Let us know if this does not address the issue.

sagarganachari, 2020-12-28:
Thanks marcschier ,,i have json file in the edge like this

  {
    "EndpointUrl": "opc.tcp://192.168.10.230:4840",
       "OpcNodes": [
      {
        "Identifier": "ns=4,i=20",
        "OpcSamplingInterval": 2000,
        "OpcPublishingInterval": 2000,
        "DisplayName": "Temperature",
        "HeartbeatInterval": 2
        
       }
    ]
  }

and in container create option 
OPC Publisher container create option
{
    "Hostname": "opcpublisher",
    "Cmd": [
        "--pf=c:/appdata/publishednodes.json",
        "--aa",
        "--di=60",
        "--to",
        "--si=0",
        "--ms=0"
    ],
    "HostConfig": {
        "Binds": [
            "c:/iiotedge:c:/appdata"
        ],
        "ExtraHosts": [
            "opctestsvr:192.168.10.230"
        ]
    }
}
OPC Publisher service is running in Edge 
Please let us know what is missing

monikavar, 2021-01-04:
Here is the duplicate issue : https://github.com/Azure/iot-edge-opc-publisher/issues/280

closed 2021-01-04 by monikavar

sagarganachari, 2021-01-04:
its not closed yet still i am facing the issue @monikavar