Azure/Industrial-IoT #996: OPC TWIN Method call

ma3goul, 2021-01-07:
**Describe the bug**
I am trying to call OPC ua method using an Opc twin module deployed on a raspberry3 , The OPC UA server is running on a siemens S7-1500 PLC . On the OPC UA server I developed a simple function block that has a struct output containing a bool , an integer and an DTL : I succeeded to call this method using the twin microservices.
But when I tried to set an output : struct of struct in the method the OPC twin module failed to connect to the endpoint and  I got this error .
What are the type of data supported by the OPC twin ?  

**Expected behavior**
Connect to the endpoint and call the method

**Screenshots**
here the twin module logs :  

[15:23:59 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Error) Skipped the type definition of "getDataFromJIG"."dataArray". Retry in next round. DataTypeNotFoundException 'Exception of type 'Opc.Ua.Client.ComplexTypes.DataTypeNotFoundException' was thrown.'
Opc.Ua.Client.ComplexTypes.DataTypeNotFoundException: Exception of type 'Opc.Ua.Client.ComplexTypes.DataTypeNotFoundException' was thrown.
   at Opc.Ua.Client.ComplexTypes.ComplexTypeSystem.AddStructuredType(IComplexTypeBuilder complexTypeBuilder, StructureDefinition structureDefinition, QualifiedName typeName, ExpandedNodeId complexTypeId, ExpandedNodeId binaryEncodingId, ExpandedNodeId xmlEncodingId)
   at Opc.Ua.Client.ComplexTypes.ComplexTypeSystem.LoadBaseStructureDataTypes(IList`1 serverStructTypes)
[15:23:59 ERR Microsoft.Azure.IIoT.OpcUa.Protocol.Services.StackLogger] (Error) Skipped the type definition of "getDataFromJIG"."readTime". Retry in next round. DataTypeNotFoundException 'Exception of type 'Opc.Ua.Client.ComplexTypes.DataTypeNotFoundException' was thrown.'
Opc.Ua.Client.ComplexTypes.DataTypeNotFoundException: Exception of type 'Opc.Ua.Client.ComplexTypes.DataTypeNotFoundException' was thrown.
   at Opc.Ua.Client.ComplexTypes.ComplexTypeSystem.AddStructuredType(IComplexTypeBuilder complexTypeBuilder, StructureDefinition structureDefinition, QualifiedName typeName, ExpandedNodeId complexTypeId, ExpandedNodeId binaryEncodingId, ExpandedNodeId xmlEncodingId)
   at Opc.Ua.Client.ComplexTypes.ComplexTypeSystem.LoadBaseStructureDataTypes(IList`1 serverStructTypes)

Thank you.

cristipogacean, 2021-01-08:
Custom structure datatypes are generally supported in the opc twin. From the exception it looks like there is a issue loading the dictionaries from the specific OPC UA server 
We have seen issues in the past with the complex types dictionaries of S7-1500. Would you have a chance to update the firmware on the S7-1500?

sagarganachari, 2021-01-11:
@ma3goul  S71500 as OPC UA server able to send data tor IOT hub using OPC Publisher,

But now i need send command from IoT hub to OPC UA server S71500 PLC, I configured OPC Twin Module but how this can be achieved......

cristipogacean, 2021-01-13:
@ma3goul would you be able to share a bit more part of the logs so we can get a bit more context on this issue?

marcschier, 2023-05-19:
Release 2.9.0 preview 2 of OPC Publisher was just released. It supports the OPC Twin API which can be invoked without cloud services.  Documentation will be provided soon.

closed 2023-05-19 by marcschier