Azure/Industrial-IoT #1004: Configured DataSetWriterId ist not applied

Haeavar, 2021-01-13:
If I configure the DataSetWriterId in the publishednodes.json, this is not applied to the publisher _DataSetMessage_ property _DataSetWriterId_ (PubSub format). Same issue for the DataSetWriterGroup. I would assume, that this will be applied to the _DataSetWriterGroup_ of the _NetworkMessage_.

This is my publishednodes.json configuration:
```
[
  {
    "EndpointUrl": "opc.tcp://*****:4840",
    "UseSecurity": false,
    "DataSetWriterId": "VRA01OPCUA1",
    "DataSetWriterGroup": "VRA01OPCUA1",
    "OpcNodes": [
      {
        "
```

These are my container create options of the publisher module:
`\"Cmd\":[\"PkiRootPath=./pki\",\"--pf=/appdata/publishednodes.json\",\"--fd=true\",\"--ll=Information\",\"--mm=PubSub\",\"--fm=true\",\"--lf=/appdata/publisher.log\",\"--aa\"]`

jehona-m, 2021-01-21:
@Haeavar  which version are you using?

Haeavar, 2021-01-22:
@jehona-m we are using the version "mcr.microsoft.com/iotedge/opc-publisher:2.7.199"
As additional info. The pubsub message format looks like following the. Where the DataSetWriterId contains the EndpointUrl from the above configuration (without the port :4840).
`{
  "body": [
    {
      "MessageId": "88418",
      "MessageType": "ua-data",
      "PublisherId": "Standalone_vra01msiot5_publisher",
      "DataSetClassId": null,
      "Messages": [
        {
          "DataSetWriterId": "opc.tcp://****_36748ee7ee3c69bd655a162d44638ac18793937d",
          "SequenceNumber": 88418,
          "MetaDataVersion": {
            "MajorVersion": 1,
            "MinorVersion": 0
          },
`

koepalex, 2021-01-22:
@Haeavar the bug has been fixed on the main branch. You can try the preview build `2.8.142-alpha` to verify that it is working as expected.

Haeavar, 2021-01-25:
@koepalex someone has removed all non-tagged versions from https://mcr.microsoft.com/v2/iotedge/opc-publisher/tags/list .
How can I pull the version `2.8.142-alpha`?

koepalex, 2021-01-27:
@Haeavar as you already figured the version was removed from mcr. There will be in the future no pre-releases (from this repository) on mcr anymore. Therefor you need either to build main branch and test it locally or for production wait for the next release (from main branch).

koepalex, 2021-02-03:
Closed due to inactivity, if main version still not working please reopen

closed 2021-02-03 by koepalex