Azure/Industrial-IoT #1005: OPC UA datatypes can not be decoded with the pubsub json format

Haeavar, 2021-01-13:
**Describe the bug**
If we subscribe e.g. the _ServerStatus_ node (i=2256) of an OPC UA server we can't decode its value using the _JsonDecoderEx_ implementation.

`                    var decoder = new JsonDecoderEx(input.Body.AsStream());
                    while (decoder.ReadEncodeable(null, typeof(NetworkMessage)) is NetworkMessage message)
...
`

The value of the corresponding payload entry for this node is _NULL_.

**To Reproduce**
Steps to reproduce the behavior:
1. Subscribe the node _"Id": "nsu=http://opcfoundation.org/UA/;i=2256"_ of an OPC UA server via publishednodes.json
2. Create an azure function with an EventHubTrigger for the incoming events
3. Deode the event data using the _JsonEncoderEx_
4. See the value of the "nsu=http://opcfoundation.org/UA/;i=2256" payload is NULL

**Expected behavior**
I would expect, any other value than NULL. 

Please advice, how to work with the OPC UA PubSub json formatted event data of the publisher in an appropriate way.

barnstee, 2021-01-15:
I can confirm that this is fixed in the latest version of OPC Publisher currently under development and produces the following JSON pyload for node ID 2256 (ServerStatus):
"ServerStatus": {
  "Value": {
    "Type": 22,
    "Body": {
      "TypeId": {
        "Id": 862
      },
      "Body": {
        "StartTime": "2021-01-15T11:30:18.4093264Z",
        "CurrentTime": "2021-01-15T11:30:24.1766754Z",
        "State": 0,
        "BuildInfo": {
          "ProductUri": "https://github.com/Azure/iot-edge-opc-plc.git",
          "ManufacturerName": "Microsoft",
          "ProductName": "IoTEdge OPC UA PLC",
          "SoftwareVersion": "1.4.363.107+a837774969718713af74ec66d8890e6fc131ddf1",
          "BuildNumber": "1.4.363.107",
          "BuildDate": "2020-10-09T13:23:30Z"
        },
        "SecondsTillShutdown": 0
      }
    }
  },
  "SourceTimestamp": "2021-01-15T11:30:24.1766754Z",
  "ServerTimestamp": "2021-01-15T11:30:24.1767558Z"
}

Until this version is released (sometime this spring), as a workaround, please subscribe to leaf nodes, i.e. the nodes with ID 2257, 2258, etc.

koepalex, 2021-08-19:
Changed to feature request, the decoding can only work with reversible encoded payload and currently the whole payload is encoded non-reversible.

marcschier, 2023-01-17:
In 2.9.0 with support for metadata messages and reversible encoding decoding using both to original OPC UA should be possible.  Recommend using UADP as encoding for compact wire format and chunking.

closed 2023-01-17 by marcschier