Azure/Industrial-IoT #1007: The deployment script of 2.7.206 doesn't work

mitkox, 2021-01-13:
**Describe the bug**
The deployment script for 2.7.206 doesn't work when following the insutrctions. 

**To Reproduce**
Steps to reproduce the behavior:
1.run ./deploy.sh -version 2.7.206

**Expected behavior**
The installation follows the steps

**Screenshots**
root@IoTEdge:/home/AzureUser/dev/Industrial-IoT# ./deploy.sh -version 2.7.206
Using '2.7.206' version...
Exception: /home/AzureUser/dev/Industrial-IoT/deploy/scripts/deploy.ps1:246
Line |
 246 |  …             throw "This script requires *git* to be installed and mus …
     |                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     | This script requires *git* to be installed and must be run from within a branch of the Industrial IoT repository. See the deployment documentation at https://github.com/Azure/Industrial-IoT to learn about other deployment options.

root@IoTEdge:/home/AzureUser/dev/Industrial-IoT#

**Desktop (please complete the following information):**
 - OS: Ubuntu Linux

**Additional context**
The deployment script is expected to work.

marcschier, 2021-01-15:
Hi @mitkix, did you download the release.zip or run the deploy.sh file from a cloned git repo?   We only support the latter right now, but are working on the former.   If the latter is what you are using, I will mark this as bug, if not, capture it as part of upcoming release.

mitkox, 2021-01-15:
Hi @marcschier I have ran deploy.sh from the cloned repo following the deployment guide steps. 

root@IoTEdge:/home/AzureUser/devs# git clone -b 2.7.206 https://github.com/Azure/Industrial-IoT
Cloning into 'Industrial-IoT'...
remote: Enumerating objects: 378, done.
remote: Counting objects: 100% (378/378), done.
remote: Compressing objects: 100% (240/240), done.
remote: Total 46979 (delta 214), reused 238 (delta 117), pack-reused 46601
Receiving objects: 100% (46979/46979), 67.42 MiB | 7.40 MiB/s, done.
Resolving deltas: 100% (31770/31770), done.
Note: switching to 'f16e7015dd681468c86a189f03ad2d2d636cf0cd'.

root@IoTEdge:/home/AzureUser/devs# cd Industrial-IoT/
root@IoTEdge:/home/AzureUser/devs/Industrial-IoT# chmod a+x deploy.sh
root@IoTEdge:/home/AzureUser/devs/Industrial-IoT# ./deploy.sh -version 2.7.206
Using '2.7.206' version...
Exception: /home/AzureUser/devs/Industrial-IoT/deploy/scripts/deploy.ps1:246
Line |
246 |  …             throw "This script requires *git* to be installed and mus …
     |                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     | This script requires *git* to be installed and must be run from within a branch of the Industrial IoT repository. See the deployment
     | documentation at https://github.com/Azure/Industrial-IoT to learn about other deployment options.

achu152, 2021-01-20:
Hi Team , 
Facing a similar issue as stated above. can you please let us know which should be the latest stable version to be used
we were using 2.7.130 before . however the version doesn't seem to be available now.

Az installed.
AzureAD installed.
Using '2.7.206' version...
This script requires *git* to be installed and must be run from within a branch of the Industrial IoT repository. See the deployment documentation at
https://github.com/Azure/Industrial-IoT to learn about other deployment options.
+ ...             throw "This script requires *git* to be installed and mus ...
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : OperationStopped: (This script req...oyment options.:String) [], RuntimeException
    + FullyQualifiedErrorId : This script requires *git* to be installed and must be run from within a branch of the Industrial IoT repository. See the deployment docum
   entation at https://github.com/Azure/Industrial-IoT to learn about other deployment options.

karok2m, 2021-01-20:
@mitkox, @achu152, please also make sure that you run the script from `release/2.7.206` branch. So you would have to run the following command from within the repo before starting the deployment:

```bash
git checkout release/2.7.206
```

achu152, 2021-01-20:
@karok2m thanks !. It works now

achu152, 2021-01-20:
@karok2m . The services got deployed as expected . However i received below msg to add the urls manually . can you please let me know where exactly these needs to be configured ? was this part to be executed by the script automatically 

###########################################################
Registering reply urls for 92702c71-536b-4968-8c7b-8669f3ab9ec6...
Insufficient privileges to complete the operation.

Registering reply urls failed. Please add the following urls to
the web app '92702c71-536b-4968-8c7b-8669f3ab9ec6' manually:
############################################################

karok2m, 2021-01-21:
@mitkox, does selecting the `release/2.7.206` branch solve the issue for you as well ? Or are you seeing some other errors ?

mitkox, 2021-01-21:
@karok2m yes thank you it works! Please update the deployment guide with the checkout step. I got two more customers failing in the same issue.

karok2m, 2021-01-27:
I've updated the [documentation](https://github.com/Azure/Industrial-IoT/blob/release%2F2.7.206/docs/deploy/howto-deploy-all-in-one.md) for `release/2.7.206` branch to mention cloning the repo using the release branch.

closed 2021-01-27 by karok2m