Azure/Industrial-IoT #1011: How do I set a node subscription to publish on data change?

gregeva, 2021-01-14:
**Is your feature request related to a problem? Please describe.**
The classic approach for most uses in a same facility is to poll data on set intervals.  When we move to geographically distributed supervision where latency, bandwidth, and limited bytes sent becomes a constraint then publish on data change becomes the option of choice.

I've only found how to create node subscriptions using a time based scan and publish rate ([documentation](https://github.com/Azure/Industrial-IoT/blob/master/docs/api/publisher/definitions.md#publisheditemapimodel)).  **How do I subscribe to a node with certain publish on data change settings?**

**Describe the solution you'd like**
I would like to be able to have the OPC Publisher scanning the local OPC UA server/controller on a specified interval, and then to publish the data upstream only when a particular data change condition was met.  For example a boolean/alarm changing state, or an analog sensor value changing by a percentage or numerical amount.

**Describe alternatives you've considered**
Various communications mechanisms allow for configuring this already - this is built into our software already.  I am trying to understand how to apply this communications pattern in the Azure Industrial IoT context.

**Additional context**
[OPC UA specification on handling DataChangeFilter and triggering DataChange Notification](https://reference.opcfoundation.org/v104/Core/docs/Part4/7.17.2/).

barnstee, 2021-01-15:
Triggering on data changes is the default for OPC Publisher for published nodes, so this is enabled already. If you additionally want data that hasn't changed on set intervals, this is done by configuring a Heartbeat.

closed 2021-01-15 by barnstee

gregeva, 2021-01-15:
Thanks @barnstee - could you clarify what you mean by configuring a heartbeat?  I can't imagine that if I had a 10 second heartbeat tag that it would also publish current values from the other tags which have not changed.

barnstee, 2021-01-15:
See docs here: https://github.com/azure/iot-edge-opc-publisher#configuration-via-configuration-file