Azure/Industrial-IoT #1014: Cloud to edge ,edge to cloud telemetry transfer dash board options

sagarganachari, 2021-01-17:
Presently I setup OPCUA Server communication with Edge device and sending data to IoT Hub, Displaying data in Power BI using Stream analytics.

Now we have requirement of both way transfer of OPCUA data points with edge and IoT Hub .
To visualize ,control the variables of  opcua in IoT Hub,for that which dashboard we can use.

OPC UA Server <<<>>>>>Edge Device<<<<<>>>>>>IoT Hub<<<<>>>>>>Dash boad

barnstee, 2021-01-19:
Have a look at the Twin Module, it allows 2 way communication.

closed 2021-01-19 by barnstee

sagarganachari, 2021-01-19:
> Have a look at the Twin Module, it allows 2 way communication.

To visualize ,control the variables of opcua in IoT Hub,for that which dashboard we can use????

barnstee, 2021-01-19:
https://github.com/Azure/Industrial-IoT/blob/master/docs/tutorials/tut-publish-data.md