Azure/Industrial-IoT #1018: IoT Edge publisher Issue

aleducq, 2021-01-20:
Discover module and Twin module are OK (can add application, publish node and read value through twin module)
publisher is running but don't upload telemetry data.

[image]

Issue found on industrial iot edge in a DMZ with firewall and proxy using 443 to communicate with Azure (Factory on the schema)
 -> first idea was that the proxy was the cause of the problem
 -> But we face the same result with iot edge in a VM  outside an industrial network without proxy (Test PC on the schema)
 -> Simulated temperature sensor works fine (we can see telemetry in the iot edge Local and Industrial

Version:
azure iot-edge 1.0.10
modules 2.7.182

Steps to reproduce the behavior:
1. Install IoT Edge
2. Add application (using postman REST API)
3. Publish Node (using postman REST API)

We should see telemtry in publisher module.

We already open a case at microsoft support (n°121011125003373 )
There s also a IcM Ticket #223097295

Regards,

aleducq, 2021-01-21:
Just adding information...
When I want to monitor to the iothub using VSCode, I can see the 2 iotedges, connected and modules are running.
The endpoints application are green meaning they are connected too
But the DataSetWriterV2 endpoints are not connected
[image]

In the publisher logs (with log level = debug) I can see messages saying that no job are requested:
```
[08:19:39 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Try querying available job...
[08:19:39 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Try querying available job...
[08:19:39 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker: aleducqiotedge_publisher_4, no job received, wait 00:00:05 ...
[08:19:39 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker: aleducqiotedge_publisher_3, no job received, wait 00:00:05 ...
[08:19:40 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Try querying available job...
[08:19:40 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Try querying available job...
[08:19:40 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker: aleducqiotedge_publisher_2, no job received, wait 00:00:05 ...
[08:19:40 DBG Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker: aleducqiotedge_publisher_1, no job received, wait 00:00:05 ...

```
And yet I have the publish nodes (using postman):
{
    "items": [
        {
            "nodeId": "nsu=KEPServerEX;s=Simulation Examples.Functions.Ramp4",
            "publishingInterval": "00:00:01",
            "samplingInterval": "00:00:00.5000000"
        },
        {
            "nodeId": "nsu=KEPServerEX;s=Simulation Examples.Functions.Ramp5",
            "publishingInterval": "00:00:01",
            "samplingInterval": "00:00:00.5000000"
        },
        {
            "nodeId": "nsu=KEPServerEX;s=Simulation Examples.Functions.Ramp6",
            "publishingInterval": "00:00:01",
            "samplingInterval": "00:00:00.5000000"
        },
        {
            "nodeId": "nsu=KEPServerEX;s=Simulation Examples.Functions.Ramp7",
            "publishingInterval": "00:00:01",
            "samplingInterval": "00:00:00.5000000"
        },
        {
            "nodeId": "nsu=KEPServerEX;s=Simulation Examples.Functions.Ramp1",
            "publishingInterval": "00:00:01",
            "samplingInterval": "00:00:00.5000000"
        },
        {
            "nodeId": "nsu=KEPServerEX;s=Simulation Examples.Functions.Ramp2",
            "publishingInterval": "00:00:01",
            "samplingInterval": "00:00:00.5000000"
        },
        {
            "nodeId": "nsu=KEPServerEX;s=Simulation Examples.Functions.Ramp3",
            "publishingInterval": "00:00:01",
            "samplingInterval": "00:00:00.5000000"
        },
        {
            "nodeId": "nsu=KEPServerEX;s=Simulation Examples.Functions.Ramp8",
            "publishingInterval": "00:00:01",
            "samplingInterval": "00:00:00.5000000"
        }
    ]
}

karok2m, 2021-01-25:
@aleducq, can you please try the same setup with our latest release, `2.7.206` ?

aleducq, 2021-01-27:
Hello @karok2m , In fact we couldn't use this version because the only release validated by PTC for OPC UA encapsulation is the `2.7.182`...
By curiousity, I've tried the `2.7.199` (which is in the approval process on PTC side) on the test PC environment and it seems OK. The gateway communicate only on 443 port.

Regarding the DataSetWriter connection issue, it was due to the fact that the `JobOrchestratorUrl` property was empty despite the fact that we've setted it correctly in the `aiiot.yaml`. Restarting the pod `aiiot-sync` has correct this and the connection has been made correctly.

closed 2021-01-27 by aleducq