Azure/Industrial-IoT #1019: Failed to launch the app with error :Storage configuration is missing in your configuration for dataprotection to store all keys across all instances.

prtest01, 2021-01-21:
Failed to launch the app with error :Storage configuration is missing in your configuration for dataprotection to store all keys across all instances.

**Steps to reproduce the behavior:**
1. git clone https://github.com/Azure/Industrial-IoT
2. cd to "master/samples/src/Microsoft.Azure.IIoT.App/src"
3. dotnet publish
4. launch Microsoft.Azure.IIoT.App.exe
5. open "https://localhost:5001" by edge

**Expected behavior**
launch successfully.

**Actually behavior**
i get this error:
[13:38:25 DBG] No keyvault uri found in configuration under PCS_KEYVAULT_URL.
[13:38:25 DBG] No keyvault uri found in environment under PCS_KEYVAULT_URL. Not reading configuration from keyvault without keyvault uri.
[13:38:26 FTL] Host terminated unexpectedly
Microsoft.Azure.IIoT.Exceptions.InvalidConfigurationException: Storage configuration is missing in your configuration for dataprotection to store all keys across all instances.

**Desktop**
OS: Windows 10 RS5
CPU: X64
.NET Build Number: 6.0.100-alpha.1.21064.42

**so where do we get these keys?**

BrentHumphreys, 2021-01-21:
The solution expects several configuration parameters.  You can pass these any number of ways, but environment variable is one common means to do it.   You will need to provision the necessary Azure infrastructure to do this as well.

jiangzeng01, 2021-01-22:
@BrentHumphreys Thanks for suggestion

marcschier, 2021-01-27:
Hi @CancanTang - Deploying the app directly to App service is not supported.   It requires the micro services and dependent Azure services to be deployed.  The deployment of the solution is documented in the docs in this repo.   If you want to run locally you can follow the steps [here](https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-local.md).  There are also instructions on how to deploy the entire set of services and the application to Azure here (https://github.com/Azure/Industrial-IoT/blob/master/docs/deploy/howto-deploy-all-in-one.md).

closed 2021-01-27 by marcschier