Azure/Industrial-IoT #1024: PublishedNodesJobConverter reads only the first subscription in publishednodes.json

Haeavar, 2021-02-02:
**Describe the bug**
My publishednodes.json contains two subscriptions to two different OPC UA server. But only the first ony will be subscribed. Is something with my publishednodes.json misconfigured?

This is my publishednodes.json:
[publishednodes.zip](https://github.com/Azure/Industrial-IoT/files/5909539/publishednodes.zip)

This is the logfile:
[publisherlog.zip](https://github.com/Azure/Industrial-IoT/files/5909553/publisherlog.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Start publisher module with attached publishednodes.json
2. Observe, that only the first subscription definition is applied (see attached logfile)

**Expected behavior**
Both subscription (all) of the publishednodes.json should be applied

**Additional context**
publisher version: 2.7.199

dacolgit, 2021-02-12:
Tested with version 2.7.199. The issue does not depend on the number of nodes in each subscription.
Ii only happens when the publisher is executed in the IoT edge context.
the following change fix also this issue:
https://github.com/Azure/Industrial-IoT/commit/3d9d503d665c1171323f4cdff52505fb5cd71c81

derSchtefan, 2021-03-04:
@dacolgit: Was this change already merged into master, and if so, when can we expect a new Docker container on the CR with these fixes?

oliverchristen, 2021-03-30:
@Haeavar  Could it be the same problem? https://github.com/Azure/Industrial-IoT/issues/669
This is not released yet. Should be part of v2.8.x for which no ETA is defined yet.

koepalex, 2021-08-17:
2.8.0 was released, that should solve this issue. Please open the Issue if error still occurs.

closed 2021-08-17 by koepalex