Azure/Industrial-IoT #1026: OPC Twin not mapping custom node namespace

maalliet, 2021-02-03:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

marcschier, 2021-02-06:
Does it work when using http://si_main_plc#i=1 instead of http://si_main_plc/#i=1?  Also, could you browse the namespaces node on the server object and let us know the namespaces uri for index 4?

maalliet, 2021-02-09:
Hi,

Browsing to both nodes _http://si_main_plc#i=1_ and _http://si_main_plc/#i=1_ return the _BadNodeIdUnknown_ error message.

I can find the namespace in the NamespaceArray node:

[image]

The Namespace node itself does not contain http://si_main_plc namespace:

[image]

I checked with the OT team and they told me that the OT setup is as follows:
There is a machine PLC that sends data over OPC UA to the main OPC UA server (which I'm connected to)
The main OPC UA server makes the data available

marcschier, 2023-06-25:
A lot of time has passed, could you retry with OPC Publisher 2.9.0-preview3 or higher and see if you can reproduce?  Also, please try against UAExpert as well to compare and please let us know. I will close for now, but please re-open if you have had a chance or are still interested in this bug.

closed 2023-06-25 by marcschier