Azure/Industrial-IoT #1036: OPC UA Publisher fails to (re)connect and gives Retry_Expired

MileBuurmeijer, 2021-02-17:
**Describe the bug**
The OPCUA Publsher module fails to recreate the Azure connection. It was good at some point but after configuration changes it failed. Redeploying the IoTEdge did not help. 

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy the OPCUA Module in your IoT Edge device contiainer runtime (running on Linux box)
2. Wait for it to startup
3. Check its log and see what the status is of IoTFactory
4. Then see error:  0: Module <my-module-id>_OPCPublisher disconnected due to Retry_Expired - now Disconnected... 
5. This error is a generic Azure connection error (see [here](https://docs.microsoft.com/en-us/dotnet/api/microsoft.azure.devices.client.connectionstatuschangereason?view=azure-dotnet)). It suggest that the retry policy (?!) was too short to succeed. 

**Expected behavior**
I expected that after restarting the docker container with the OPC UA publisher module it would reconnect to Azure without problem.

**Logfile**
[19:04:45 INF Root] Starting module OpcPublisher version 2.7.206.36449. 
[19:04:45 INF Root] Initiating prometheus at port 9702/metrics 
[19:04:45 INF Root] Prometheus metric server started. 
[19:04:45 WRN Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Bypassing certificate validation for client. 
[19:04:45 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] Running in iotedge context. 
[19:04:45 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Module xxx_OPCPublisher disconnected due to Retry_Expired - now Disconnected... 

**Desktop (please complete the following information):**
 - OS: Linux Ubuntu 20.10

**Azure module config file**
{
   "Hostname":"publisher",
   "Cmd":["publisher",
   "--pf=/appdata/publishednodes.json",
   "--di=60",
   "--tm",
   "--aa",
   "--si=0",
   "--ms=0"],
   "ExposedPorts":{"62222/tcp":{}},
   "HostConfig":
       {
               "Binds": ["/data:/appdata"],
	       "ExtraHosts": [ "<localhostname>:127.0.0.1" ]
       }
}
.

logg-sar, 2021-02-25:
@MileBuurmeijer 

Hi, have you changed the hostname between installing the packages and running the service?

Seems to be the same Issue like [https://github.com/Azure/iotedge/issues/4466]
I found a workaround - it may help you too.

koepalex, 2021-03-02:
Please check, that there is not an second OPC Publisher instance is running with the same connection string, this is what typically cause this issue. For analysis please provide the publishednodes.json and the full log.

koepalex, 2021-03-11:
@MileBuurmeijer does the issue still exist? If so please add more context like the description of what kind of `configuration changes` (until end of next week).

closed 2021-03-19 by marcschier