Azure/Industrial-IoT #1041: OPC Twin arm64 image not working

MatthijsvdVeer, 2021-02-25:
**Describe the bug**
I'm running an arm64 edge device with Azure IoT Edge 1.1.0. When I run the OPC Twin module (version 2.7.183-linux-arm64v8 or version 2.7.199-linux-arm64v8) the module fails with this error:
`standard_init_linux.go:219: exec user process caused: no such file or directory`

A quick search online suggests that this has to do with some line endings in the files included in this image. Also, just pulling :latest wouldn't result in an arm64 image, hence the specific version.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a deployment with a module with image `mcr.microsoft.com/iotedge/opc-twin:2.7.183-linux-arm64v8` or `mcr.microsoft.com/iotedge/opc-twin:2.7.199-linux-arm64v8`
2. Deploy to Azure IoT Edge device running IoT Edge 1.1.0
3. Check the logs of the module
4. `standard_init_linux.go:219: exec user process caused: no such file or directory`

**Expected behavior**
A running module

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - Architecture: arm64v8

marcschier, 2021-02-26:
Hi, can you try "mcr.microsoft.com/iotedge/opc-twin:2.7.206"?  Also, assumption is that this is running on a Raspberry Pi, correct?

MatthijsvdVeer, 2021-02-26:
Hi Marc, that doesn't work, same error. This is not running on a Raspberry Pi, but on a production gateway.

marcschier, 2021-07-14:
Please try again with 2.8.0 release.

closed 2021-07-14 by marcschier