Azure/Industrial-IoT #1042: Connectivity for PLC that are not OPC UA

kbeaugrand, 2021-02-28:
**Is your feature request related to a problem? Please describe.**
Would like to provide this kind of features for PLC that are not under an OPC UA gateway..

**Describe the solution you'd like**
I'm working in an organization that could not always leverage on OPC Gateway to provide connectivity to theire machines.
They would also not like to retro-fit their plant machines network (due to costs, delay, complexity, ...) to add this kind of gateaway.

**Describe alternatives you've considered**
Does that make sens to you ?
If true, I can take time to implement thoses features on some well known PLC like (Siemens S7, ControlLogix, ...) to get started.

aagamb12, 2021-02-28:
One thing we have to done to overcome touching, upgrading legacy Automation control systems (DCS, BMS, SCADA etc)) and stanadalone pieces of equipment (PLC &HMI )is to deploy a cheap, non intrusive, agnostic IoTGateway as another listening node in the control systems architecture that can collect tags information directly from the PLCs or controllers (Rockwell, Siemens, Honeywell  etc )and then send the data to the cloud successfully. There are many inexpensive IoTgateways in the market right now,

kbeaugrand, 2021-02-28:
Hi @aagamb12, 
Thank you for your answer. 

I'm looking for some IoT Gateway that could bring us the possibility offered by Azure IoT Hub such: 

- Edge Analytics, 
- Central configuration of many gateways,
- Custom Edge Module development, 
- Custom Data destination (to fullfill security requirements for some cases), 
- ...

However I prefered to leverage on only 1 kind of IoT platform to support our cases that cames from standard LoRa devices to PLC. 
The possibility to brign our assets to Digital Twins is clearly interesting. 
Unfortunatly, I wasn't able to find an industrial IoT Gateway that can give us thoses features through Azure IoT Hub (That I'm currently using).
For that reason I already have developed some IoT Edge Modules that can connect on PLC (actually Rockwell and Siemens), and periodically send some data throught IoT Edge. In that case, we imagin deploying that with some Industrial (Raspberry's like) hardwares.
This kind of deployment is also cheap, but kindly intrusive where machines are not connected on the plant network (that's also why we cannot leverage on usual Gateways presenting OPC UA).

Maybe I'm wrong in my expectations or assessments that's why I prefer to ask you your sentiment about this...

What kind of IoT Gateway (inexpensive and not intrusive) have you in mind to provide connectivity to the IoT Edge capabilities have you in mind ?

kbeaugrand, 2021-03-06:
Up ?

koepalex, 2021-03-09:
@jehona-m could you follow-up on this request ?

closed 2023-01-16 by marcschier