Azure/Industrial-IoT #1048: New-ADApplications in aad-register.ps1 can return null

WilliamBerryiii, 2021-03-02:
**Describe the bug**
In `aad-register.ps1` the `New-ADApplications` function can return null if if the `$creds` variable has not been set.  See [this line](https://github.com/Azure/Industrial-IoT/blob/b14ebd3aa3a7464477cfb1499cfa3fabd788310e/deploy/scripts/aad-register.ps1#L230).
This null value is not checked before being piped to `ConvertTo-Json` and the script exits without an insightful warning to the user.

**Expected behavior**
A useful error is returned to the user if the collection of credentials is unsuccessful. This is partially mitigated by PR #1046

koepalex, 2021-03-31:
@WilliamBerryiii can you describe if something is open for your issue, or this issue can be closed?

WilliamBerryiii, 2021-03-31:
@koepalex - I believe this can now be resolved. The PR looks to be merged.

closed 2021-03-31 by WilliamBerryiii