Azure/Industrial-IoT #1053: Username/password credential support for OPC Twin?

MatthijsvdVeer, 2021-03-03:
**Is your feature request related to a problem? Please describe.**
We're trying to leverage the OPC Twin Edge Module to write values to OPC UA servers. We don't want to deploy the entire IIOT solution, instead, we want to use the edge module in combination with our own platform. The servers in question use username/password security. I can't find any documentation about the support of this scenario. Is this supported?

**Describe the solution you'd like**
I want to use the write (and possibly read) functionality of the OPC Twin to endpoints that support username/password credentials.

**Describe alternatives you've considered**
For publishing values, we currently use OPC Publisher, it supports username/password validation.

ajithkumars, 2021-03-26:
I have the same question, does opc twin not check pn.json or acquire this from publisher?.
How to support this? Also, if certificates are used?

mwiot, 2021-04-06:
Hi, 
I have the same question as MatthijsvdVeer, and hope someone can give a quick reply regarding this issue. 

1. is it possible to use "Username/Password" as authentication method for the twin module, or is the only supported method "Anonymous"? And if possible, how?
2. We are using the entire Industrial-IoT platform. Can username/passsword authentication method be specified through REST microservice?

mwiot, 2021-04-09:
Hi again, 

this can be solved by using a json body like in the following example.  

/twin/v2/write/{endpointId}

`{
   "nodeId":"nsu=mynamespace;s=mynode",
   "value":15.7,
   "header":{
      "elevation":{
         "type":"UserName",
         "value":{
            "user":"myuauser",
            "password":"myuapassword"
         }
      }
   }
}`

hansgschossmann, 2021-04-13:
Thanks @mwiot for providing the information. 

Closing it for now.

closed 2021-04-13 by hansgschossmann