Azure/Industrial-IoT #1056: Signing in process for a deployment takes considerably long

NooraKubati, 2021-03-04:
**Describe the bug**
When trying to deploy version 2.7 on Power Shell, the initial stages are ok but once it gets to 'Signing in...' it sticks for quite a while. The length of time has varied for us from 5mins - 2hours. On average it can take about 20 minutes to sign in. Note: This is just waiting for the sign in window to even appear, before credentials have been entered. 

**To Reproduce**
Steps to reproduce the behavior:
1. Clone repo 
2. CD to Industrial IoT folder
3. Run command ".\deploy -version 2.7"

**Expected behavior**
After speaking with someone from the Microsoft team we understand this is not normal nor intended. The process should be a lot quicker. 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Windows Powershell 64bit, Windows Powershell x86

**Additional context**
Initially we thought it may be scaling in relation to the number of subscriptions on the user account, but we've seen nothing conclusive as of yet.

karok2m, 2021-03-08:
@NooraKubati, can you please pass id (`subscriptionId`) of the subscription that you would like to use or its name (`subscriptionName`) to the script and see if that accelerates it or not ? Please check [this page](https://github.com/Azure/Industrial-IoT/blob/release/2.7.206/docs/deploy/howto-deploy-all-in-one.md#deployment-script-options) for the list of deployment script options.

NooraKubati, 2021-03-12:
@karok2m, Hi I have just attempted this and the signing in process takes at the very least 5 minutes still. My understanding is that it should be relatively instantaneous so I believe the problem still exists.

marcschier, 2021-05-07:
I ran into an issue recently where the sign in (Connect-Azure) was hanging for a long time (forever because I typically give up after a minute or two).  I tried to fix it by using --use-device-code instead which produced a Az.Account versioning error.  In the course of addressing this versioning error the hang also resolved itself for me.  

Essentially here is what I did:

* Remove all Az and Az.* and AzureRM modules from your machine as per the official install/uninstall instructions for Az.
* Remove all remaining occurrences of Az and AzureRM modues as per the instructions here: follow https://github.com/Azure/azure-powershell/blob/master/documentation/troubleshoot-module-load.md (essentially delete the orphaned modules if you have any from your file system / module load path).
* Install latest Az using the official instructions.

hansgschossmann, 2021-07-20:
Seem that this is tno a bug, but an issue which was seen because of outdated components. Closing for now. Pls reactivate if it is seen again.

closed 2021-07-20 by hansgschossmann

saramonteiro, 2021-12-28:
This issue is also happening to me.
The script is stuck on Signing step for hours.

karok2m, 2022-01-24:
@saramonteiro, can you please go over the steps outlined by @marcschier above and see if that accelerates sign-in process for you ?

saramonteiro, 2022-01-30:
> @saramonteiro, can you please go over the steps outlined by @marcschier above and see if that accelerates sign-in process for you ?

@karok2m I got to solve it by:
Using Connect-AzAccount -TenantId <my tenant id>
We also can use az login before running the script