Azure/Industrial-IoT #1060: publishednodes.json changes, log indicates that nothing has changed and nothing will be done, but system proceeds to process changes

derSchtefan, 2021-03-11:
**Describe the bug**
When the publishednodes.json changes, the message in the log indicates that a file change was detected but the content-hash is equal to last one, and nothing will be done. The system then however continues to correctly process the changes.

`[2021-03-11 19:04:46Z INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.LegacyJobOrchestrator] File /appdata/publishednodes.json has changed and content-hash is equal to last one, nothing to do`

A full log can be found at the bottom. There  is also a message appearing that the contents of the file HAVE changed. It seems the second message is an artifact of some other change detector happening.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a publishednodes.json
2. Start OPC Publisher
3. Change the json file and observe the two entries in the log

**Expected behavior**
Log message does not appear, or log message is amended to indicate that an earlier detection has already processed the change.

**Additional context**
Version is the current master branch dcd7a4a57d6e0e25936769f1496d687626d1f282.
Since there were quite some changes to the pn.json processing recently, it is probably a better approach to slightly reword the log message.
**Full log**

**Full log**

```
[2021-03-11 19:33:28Z INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.LegacyJobOrchestrator] File /appdata/publishednodes.json has changed, last known hash 83B87FEE198B7A9DF9EEB354BE3FA734F6FF32B4CF825FC80A8E33FC0530E7BF, new hash E4B1F1E3552FEE2972BB99F27AE856088ECD63DB0A7781B433C9C9FD4B478F45, reloading...
[2021-03-11 19:33:28Z INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Read 2 items from published nodes file in 00:00:00.0130894
[2021-03-11 19:33:28Z INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Models.PublishedNodesJobConverter] Converted items to jobs in 00:00:00.0133206
[2021-03-11 19:33:28Z INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.LegacyJobOrchestrator] Job opc.tcp://192.168.1.131:4841/GFUnitServer_c117a8ecf9270b256fa746035cc6da440d64cd42 loaded. DataSetWriters endpoints: opc.tcp://192.168.1.131:4841/GFUnitServer
[2021-03-11 19:33:28Z INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.LegacyJobOrchestrator] Job opc.tcp://192.168.1.131:4840/GFUnitServer_f48c48babe63fb83e0d38da668724789a47f910a loaded. DataSetWriters endpoints: opc.tcp://192.168.1.131:4840/GFUnitServer
[2021-03-11 19:33:28Z INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.LegacyJobOrchestrator] File /appdata/publishednodes.json has changed and content-hash is equal to last one, nothing to do
[2021-03-11 19:33:31Z INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker+JobProcess] Job opc.tcp://192.168.1.131:4841/GFUnitServer_c117a8ecf9270b256fa746035cc6da440d64cd42 cancelled.
[2021-03-11 19:33:31Z INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker: StandalonePublisher_1, Job: opc.tcp://192.168.1.131:4841/GFUnitServer_c117a8ecf9270b256fa746035cc6da440d64cd42 processing completed ...
```

koepalex, 2021-03-18:
The reason is that there are more then one FileChangedNotification for the modification of one file. The first, changes the hash and will lead to changes and the second (or following) will be ignored. So technical the log message is correct.

dacolgit, 2021-09-02:
fixed in PR [1371](https://github.com/Azure/Industrial-IoT/pull/1371)

closed 2021-09-02 by dacolgit