Azure/Industrial-IoT #1074: OPCPublisher: is it possible to monitor items on DataChangeTriggerType.StatusValueTimestamp ?

apittier, 2021-03-18:
Hi!

Is it possible to monitor items on DataChangeTriggerType.StatusValueTimestamp ?

From the OPC UA Specification we have that the default behavior is to ["Report a notification if either the StatusCode or the value change. "](https://reference.opcfoundation.org/v104/Core/docs/Part4/7.17.2/) (i.e., _Microsoft.Azure.IIoT.OpcUa.Publisher.Models.DataChangeTriggerType.StatusValue_).

I was wondering if there's any mechanism available when configuring the OPCPublisher to define that we want to be notified on _SourceTimestamp_ changes also (i.e., DataChangeTriggerType.StatusValueTimestamp).

Many thanks!
Amparo

marcschier, 2021-03-19:
Hi @apittier, unfortunately this is not implemented.  We will evaluate it for our backlog.

apittier, 2021-03-19:
Thanks for answering.

marcschier, 2021-03-20:
Keeping on backlog

PR #1830 (2022-11-09): Support data change trigger configuration in node
So that the OPC Publisher can be configured to receive a notification on Status, Status/Value or Status/Value/Timestamp change.

This change adds the field `"OpcNodes[].DataChangeTrigger"` that can take the value "Status", "StatusValue" or "StatusValueTimestamp". This allows the configuration of the monitored item data change trigger. It can be left null, in which case the default defined in the OPC UA specs is "StatusValue".

Fixes #1074
commit 9e5cba5f (2022-11-09): Support data change trigger configuration in node (#1830)

* Support data change trigger configuration in node

So that the OPC Publisher can be configured to receive a notification
on Status, Status/Value or Status/Value/Timestamp change.

This change adds the field `"OpcNodes[].DataChangeFilter"` that can take
the value "Status", "StatusValue" or "StatusValueTimestamp". This allows
the configuration of the monitored item data change trigger. It can be
left null, in which case the default defined in the OPC UA specs is
"StatusValue".

Fixes #1074

*
closed 2022-11-09 by marcschier