Azure/Industrial-IoT #1077: deploy.ps1 script needs to be run from within a branch of the Industrial IoT repository

wkoschen-at-ptc, 2021-03-22:
**Describe the bug**

deploy.ps1 script needs to be run from within a branch of the Industrial IoT repository.

Recently several of our automated deployments started failing.  Upon inspection of the issue the root cause was discovered to be a release branch which was deleted  **(release/2.7.199)**.  We were informed this should be expected and to use the release tag in place of the branch.

This morning we attempted to test a deployment by checking out the corresponding release tag and the deployment failed during execution of **deploy.ps1** with the following error message:

"This script requires *git* to be installed and must be run from
 within a branch of the Industrial IoT repository.
 See the deployment documentation at https://github.com/Azure/Industrial-IoT
 to learn about other deployment options."

**To Reproduce**
Steps to reproduce the behavior:
1. Clone this repo
2. Checkout a release tag
3. Run the deploy.ps1 script
4. See error

**Expected behavior**
It is unclear what the expected behavior should be.  Should the script require a branch?  

**Additional Notes**

How can we continue to support older releases using the release tag?

marcschier, 2021-03-25:
Hi, @wkoschen-at-ptc , best would be for your team to fork (top right button) the Github repo. Clone your fork (copy) and create your desired branches there. For example a release branch on the release tag.  Make sure to name the branch release/{releasetag} for all old 2.7 releases and then run deploy.ps1 from that branch.  Please note that at this point only the releases that have release branches on Azure/IoT-Industrial are receiving support.  As of this writing this is 2.5 (2.5.4) and 2.7 (2.7.206).

closed 2021-03-25 by marcschier

payald15, 2021-04-08:
@marcschier probably this bug could be re-opened now based on our discussion in the meeting this morning. As a resolution to this issue, if a parameterized script could be provided that can replace the original repo's url with forked one then it should be good.

I followed the steps you mentioned earlier on this thread [followed them specifically for 2.7.199], and here is the observation:

1. Forked the Azure Industrial-IoT Repo : Azure/Industrial-IoT [ you can have a look here: [https://github.com/PayalDodeja/Industrial-IoT]
2. Cloned the forked repo on my system
3. Ran the command **git checkout 2.7.199**
4.  Ran the command **git branch release/2.7.199**
5.  Ran the command **git push origin release/2.7.199**
6. Now, when the deployment script was run using this command **.\deploy.ps1 -type minimum -version 2.7.199 -aadConfig aad.json** , following error popped up

> 11:16:54 AM - Error: Code=InvalidContentLink; Message=Unable to download deployment content from 'https://raw.githubusercontent.com/Azure/Industrial-IoT/release/2.7.199/deploy/templates/azuredeploy.minimum.json'. The tracking Id is '1e77a417-1a25-4e7f-831b-e2a9261556b2'. Please see https://aka.ms/arm-deploy for usage details.

7. The script is looking for the template under the release branch present at the default url's location [ that is original repo]
8. There are around 28 places where the default url is the original repo's url
9. In order to make the minimum deployment work, I had to make 2 changes in the azuredeploy.json template. They both were replacing the original repo's url with my forked repo's url

> Find the section below in azuredeploy.json file and replace "defaultUrl" under "repoUrl" and "templateUrl" with the path for your forked repo
        "repoUrl": {
            "type": "string",
            "defaultValue": "https://github.com/Azure/Industrial-IoT",
            "metadata": {
                "description": "The repository url from which to deploy services and application.  Default is official repository."
            }
        },
        "templateUrl": {
            "type": "string",
            "defaultValue": "https://raw.githubusercontent.com/Azure/Industrial-IoT",
            "metadata": {
                "description": "The artifacts url from which to pull all linked templates.  Default is official repository."
            }
            
10.  After making the change, had to push this change in upstream.
11. Now when this command was run again, **.\deploy.ps1 -type minimum -version 2.7.199 -aadConfig aad.json**  , successful minimal deployment of resources with suite version 2.7.199 was performed.

PayalDodeja, 2021-04-08:
@marcschier probably this bug could be re-opened now based on our discussion in the meeting this morning. As a resolution to this issue, if a parameterized script could be provided that can replace the original repo's url with forked one then it should be good.

I followed the steps you mentioned earlier on this thread [followed them specifically for 2.7.199], and here is the observation:

1. Forked the Azure Industrial-IoT Repo : Azure/Industrial-IoT [ you can have a look here: [https://github.com/PayalDodeja/Industrial-IoT]
2. Cloned the forked repo on my system
3. Ran the command git checkout 2.7.199
4. Ran the command git branch release/2.7.199
5. Ran the command git push origin release/2.7.199
6. Now, when the deployment script was run using this command .\deploy.ps1 -type minimum -version 2.7.199 -aadConfig aad.json , following error popped up

> 11:16:54 AM - Error: Code=InvalidContentLink; Message=Unable to download deployment content from [...]

7.The script is looking for the template under the release branch present at the default url's location [ that is original repo]
8.There are around 28 places where the default url is the original repo's url
9.In order to make the minimum deployment work, I had to make 2 changes in the azuredeploy.json template. They both were replacing the original repo's url with my forked repo's url

> Find the section below in azuredeploy.json file and replace "defaultUrl" under "repoUrl" and "templateUrl" with the path for your forked repo
> "repoUrl": {
> "type": "string",
> "defaultValue": [...]

10.After making the change, had to push this change in upstream.
11.Now when this command was run again, .\deploy.ps1 -type minimum -version 2.7.199 -aadConfig aad.json , successful minimal deployment of resources with suite version 2.7.199 was performed.