Azure/Industrial-IoT #1085: ARM32V7: Not supported anymore

dvanlaerhoven, 2021-03-30:
Hi All!

I've used the OPCPublisher module this summer with succes on an arm32v7 device. 
I believe I've done the same thing, but now I get:

```
root@axcf2152:/opt/plcnext/# docker run -it mcr.microsoft.com/iotedge/opc-publisher:latest
standard_init_linux.go:211: exec user process caused "exec format error"
failed to resize tty, using default size
``` 

Is arm32v7 not longer supported? I could not find a decapitated message.
Can I build the image myself, what base image should I take?

With Kinds regards

Dries

marcschier, 2021-04-08:
We do not publish containers explicitly tagged for arm32 and arm64, but publish instead a multi arch manifest list under the release tag (e.g., latest, 2, 2.7, 2.7.206).  This one seemed to have a problem with the arm64 and arm32 layers which we are investigating.  

One thing to try is to instead of 2.7.206 use `2.7.206@sha256:11493b39361dd38cbe27f5debb4472ae14ceb68008f7b6e8f9ba41a5f9cb2915` which references the linux/arm layer in the current 2.7.206 image list.  However, this has not been tested yet.

koepalex, 2021-04-08:
Tested with Raspberry PI 3

```bash
pi@rpi-3-edge:~/publisher-config $ uname -a
Linux rpi-3-edge 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
pi@rpi-3-edge:~/publisher-config $ dpkg --print-architecture
armhf
pi@rpi-3-edge:~/publisher-config $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
pi@rpi-3-edge:~/publisher-config $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 76.80
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 000000004f553990
Model           : Raspberry Pi 3 Model B Rev 1.2
```

I used the latest OPC Publisher as module of IoT Edge with the following container create options:

```json
{
  "Hostname": "publisher",
  "Cmd": [
    "PkiRootPath=/cfg/pki",
    "--pf=/cfg/pn.json",
    "--lf=/cfg/publisher.log",
    "--aa"
  ],
  "HostConfig": {
    "Mounts": [
      {
        "Type": "bind",
        "Source": "/home/pi/publisher-config",
        "Target": "/cfg"
      }
    ]
  }
}
```

Here the output of the running publisher:

```bash
pi@rpi-3-edge:~/publisher-config $ iotedge logs publisher -f --tail 25
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :             83 (0.1/s)
  # Connection retries                 :              0

[14:22:30 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine]
  DIAGNOSTICS INFORMATION for          : opc.tcp://172.18.0.5:50000_5a3ab012f8c29c0c70622402b8ce2b683743e5c1
  # Ingestion duration                 :    00:00:14:53 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :            983 (1.1/s)
  # Ingress ValueChanges (from OPC)    :         47,570 (53.22/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :         47,410
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :             89
  # Encoder avg Notifications/Message  :            533
  # Encoder avg IoT Message body size  :         93,287 (36%)
  # Encoder avg IoT Chunk (4 KB) usage :           22.8
  # Estimated IoT Chunks (4 KB) per day:        197,884
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :             89 (0.1/s)
  # Connection retries                 :              0
```

Please check that your docker runtime (e.g. mobi),  iot edge and rasbian are up-to-date. If the error still occurs please reopen the issue with additional information:

* uname -a
* dpkg --print-architecture
* cat /proc/cpuinfo
* iotedge --version
* docker --version

closed 2021-04-08 by koepalex