Azure/Industrial-IoT #1086: OPCPublisher module: BadMonitoredItemIdInvalid when first OpcNode has a HeartbeatInterval defined

oliverchristen, 2021-03-30:
**Describe the bug**
I run the OPCPublisher in standalone mode with pn.json configuration. When the first OpcNode has a HeartbeatInterval defined because it's value does not change often but shall be sent anyway, the Value is sent as StatusCode `BadMonitoredItemIdInvalid`

```
[
	{
		"NodeId": "http://www.softing.com/dataFEEDedge/SiemensS7/nsuri?connectionname=H2#s=DB+Compressor.CI_M_301.Compressor.HMI.O_Operating_State",
		"ApplicationUri": "urn:31e75bf9a076/Softing/dataFEED_edge_OpcUaServer",
		"DisplayName": "Compressor_RunningState",
		"Value": {
			"StatusCode": {
				"Symbol": "BadMonitoredItemIdInvalid",
				"Code": 2151809024
			},
			"SourceTimestamp": "2021-03-30T07:09:16.9330875Z"
		}
]
```

The missing closing curly bracket is on purpose. That's another problem. It's not a valid Json in this case.

When I move this OpcNode configuration to the end of the array in the pn.json then everything works as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure 49 OpcNode's. First has a HeartbeatInterval of 20. The rest hasn't. It's important to have many nodes. It's not reproducible with only a small number of nodes.
2. Restart OPCPublisher module
=> The first node shows the StatusCode `BadMonitoredItemIdInvalid` as described above and the JSON is invalid.
3. Move the OpcNode with the HeartbeatInterval to the end of the array of 49 OpcNodes.
4. Restart OPCPublisher module
=> All the node values are sent as expected. The node with the heartbeat interval is sent even if the value hasn't changed.

**Expected behavior**
1. The order of the OpcNode's does not matter. 
2. The BadMonitoredItemIdInvalid status code should not be sent when everything is configured correctly. 
3. The JSON message is a valid JSON even if there are StatusCode's as Value's.

**Version of OPCPublisher module**
2.7.206

koepalex, 2021-08-19:
I was not able to reproduce with 2.8 release, could you please share the pn.json and the iot edge support bundle or retest with 2.8 release?

oliverchristen, 2021-08-19:
I don't work with OpcPublisher any more. Thank you anyway.

closed 2021-08-19 by oliverchristen