Azure/Industrial-IoT #1088: OPC Publisher Module on a PI4 "standard_init_linux.go:219: exec user process caused: no such file or directory"

DanielBott, 2021-04-01:
I am trying to get the OPC Publisher working on a PI4 with Ubuntu 18.04. After the module is deployed, it does not start and has the following error:
```
ubun1804@Rasp4:~$ sudo iotedge logs OPCPublisher
standard_init_linux.go:219: exec user process caused: no such file or directory
```

the main iotedge modules are running:
```
ubun1804@Rasp4:~$ sudo iotedge list
NAME             STATUS           DESCRIPTION               CONFIG
OPCPublisher     failed           Failed (1) 2 minutes ago  mcr.microsoft.com/iotedge/opc-publisher:latest
edgeAgent        running          Up 4 hours                mcr.microsoft.com/azureiotedge-agent:1.0
edgeHub          running          Up 4 hours                mcr.microsoft.com/azureiotedge-hub:1.0
```

in my point of view the iotedge is running very well
```
ubun1804@Rasp4:~$ sudo iotedge check
Configuration checks
--------------------
√ config.yaml is well-formed - OK
√ config.yaml has well-formed connection string - OK
√ container engine is installed and functional - OK
√ config.yaml has correct URIs for daemon mgmt endpoint - OK
√ host time is close to real time - OK
√ container time is close to host time - OK
√ production readiness: container engine - OK
Connectivity checks
-------------------
√ host can connect to and perform TLS handshake with IoT Hub AMQP port - OK
√ host can connect to and perform TLS handshake with IoT Hub HTTPS / WebSockets port - OK
√ host can connect to and perform TLS handshake with IoT Hub MQTT port - OK
√ container on the default network can connect to IoT Hub AMQP port - OK
√ container on the default network can connect to IoT Hub HTTPS / WebSockets port - OK
√ container on the default network can connect to IoT Hub MQTT port - OK
√ container on the IoT Edge module network can connect to IoT Hub AMQP port - OK
√ container on the IoT Edge module network can connect to IoT Hub HTTPS / WebSockets port - OK
√ container on the IoT Edge module network can connect to IoT Hub MQTT port - OK
16 check(s) succeeded.
```
the setup followed https://docs.microsoft.com/en-us/azure/iot-edge/how-to-install-iot-edge?view=iotedge-2018-06 

my intallation runs since some days
```
ubun1804@Rasp4:~$ sudo docker container list
CONTAINER ID   IMAGE                                      COMMAND                   CREATED       STATUS       PORTS                                                                  NAMES
10611d7fb73f   mcr.microsoft.com/azureiotedge-hub:1.0     "/bin/sh -c 'echo \"$…"   12 days ago   Up 5 hours   0.0.0.0:443->443/tcp, 0.0.0.0:5671->5671/tcp, 0.0.0.0:8883->8883/tcp   edgeHub
8143f0b6ba86   mcr.microsoft.com/azureiotedge-agent:1.0   "/bin/sh -c 'exec /a…"    12 days ago   Up 5 hours                                                                          edgeAgent
```

also 
(1) I tried to deploy other edge device modules and it works. e.g. a demo module written by myself or the azure-iot.simulated-temperature-sensor module
(2) I also deployed other OPC Publisher Verions 2.5.3 and 2.3x. e.g. 2.5.3:
```
ubun1804@Rasp4:~$ sudo iotedge logs edgeAgent
<6> 2021-03-31 10:58:53.157 +00:00 [INF] - Plan execution ended for deployment 25
<3> 2021-03-31 10:58:53.157 +00:00 [ERR] - Edge agent plan execution failed.
System.AggregateException: One or more errors occurred. (Error calling prepare update for module OPCPublisher: Could not prepare update for module "OPCPublisher"
        caused by: Could not pull image mcr.microsoft.com/iotedge/opc-publisher:2.5.3
        caused by: no matching manifest for linux/arm64/v8 in the manifest list entries)
 ---> Microsoft.Azure.Devices.Edge.Agent.Edgelet.EdgeletCommunicationException- Message:Error calling prepare update for module OPCPublisher: Could not prepare update for module "OPCPublisher"
        caused by: Could not pull image mcr.microsoft.com/iotedge/opc-publisher:2.5.3
        caused by: no matching manifest for linux/arm64/v8 in the manifest list entries, StatusCode:500, at:   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Version_2020_07_07.ModuleManagementHttpClient.HandleException(Exception exception, String operation) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/version_2020_07_07/ModuleManagementHttpClient.cs:line 232
   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Versioning.ModuleManagementHttpClientVersioned.Execute[T](Func`1 func, String operation) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/versioning/ModuleManagementHttpClientVersioned.cs:line 143
   at Microsoft.Azure.Devices.Edge.Agent.Edgelet.Version_2020_07_07.ModuleManagementHttpClient.PrepareUpdateAsync(ModuleSpec moduleSpec) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Edgelet/version_2020_07_07/ModuleManagementHttpClient.cs:line 214
   at Microsoft.Azure.Devices.Edge.Agent.Core.Commands.GroupCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/commands/GroupCommand.cs:line 35
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 67
   at Microsoft.Azure.Devices.Edge.Agent.Core.Commands.GroupCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/commands/GroupCommand.cs:line 35
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 67
   at Microsoft.Azure.Devices.Edge.Agent.Core.PlanRunners.OrderedRetryPlanRunner.ExecuteAsync(Int64 deploymentId, Plan plan, CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/planrunners/OrdererdRetryPlanRunner.cs:line 83
[... 13 lines cut ...]
   at Microsoft.Azure.Devices.Edge.Agent.Core.Commands.GroupCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/commands/GroupCommand.cs:line 35
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 67
   at Microsoft.Azure.Devices.Edge.Agent.Core.Commands.GroupCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/commands/GroupCommand.cs:line 35
   at Microsoft.Azure.Devices.Edge.Agent.Core.LoggingCommandFactory.LoggingCommand.ExecuteAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/LoggingCommandFactory.cs:line 67
   at Microsoft.Azure.Devices.Edge.Agent.Core.PlanRunners.OrderedRetryPlanRunner.ExecuteAsync(Int64 deploymentId, Plan plan, CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/planrunners/OrdererdRetryPlanRunner.cs:line 83
   --- End of inner exception stack trace ---
   at Microsoft.Azure.Devices.Edge.Agent.Core.PlanRunners.OrderedRetryPlanRunner.<>c.<ExecuteAsync>b__7_0(List`1 f) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/planrunners/OrdererdRetryPlanRunner.cs:line 115
   at Microsoft.Azure.Devices.Edge.Agent.Core.PlanRunners.OrderedRetryPlanRunner.ExecuteAsync(Int64 deploymentId, Plan plan, CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/planrunners/OrdererdRetryPlanRunner.cs:line 115
   at Microsoft.Azure.Devices.Edge.Agent.Core.Agent.ReconcileAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/Agent.cs:line 203
   at Microsoft.Azure.Devices.Edge.Agent.Core.Agent.ReconcileAsync(CancellationToken token) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.Core/Agent.cs:line 203
```
(3) it makes no difference if I use: 
```
mcr.microsoft.com/iotedge/opc-publisher:latest or 
mcr.microsoft.com/iotedge/opc-publisher:preview
```
 i get following error:
```
ubun1804@Rasp4:~$ sudo iotedge logs OPCPublisher
standard_init_linux.go:219: exec user process caused: no such file or directory
```

Do you have any idea what it can be?

_Originally posted by @DanielBott in https://github.com/Azure/Industrial-IoT/issues/291#issuecomment-811177227_

DanielBott, 2021-04-01:
update: i found this article https://github.com/Azure/Industrial-IoT/issues/1034 and tried the version 2.7.199 as well 2.7.206

```
ubun1804@Rasp4:~$ sudo iotedge list
NAME             STATUS           DESCRIPTION                CONFIG
OPCPublisher     failed           Failed (1) 18 seconds ago  mcr.microsoft.com/iotedge/opc-publisher:2.7.206
edgeAgent        running          Up a day                   mcr.microsoft.com/azureiotedge-agent:1.0
edgeHub          running          Up a day                   mcr.microsoft.com/azureiotedge-hub:1.0

ubun1804@Rasp4:~$ sudo iotedge logs OPCPublisher
standard_init_linux.go:219: exec user process caused: no such file or directory
standard_init_linux.go:219: exec user process caused: no such file or directory
standard_init_linux.go:219: exec user process caused: no such file or directory
```

```
sudo journalctl -u iotedge --no-pager --no-full -n -f
[INFO] - Pulling image mcr.microsoft.com/iotedge/opc-publisher:2.7.206...
[INFO] - Successfully pulled image mcr.microsoft.com/iotedge/opc-publisher:2.7.206
[INFO] - [mgmt] - - - [2021-04-01 18:17:37.375084276 UTC] "POST /modules/OPCPublisher/prepareupdate?api-version=2020-07-07 HTTP/1.1" 204 No Content …"-" "-" auth_id(-)
[INFO] - Stopping module OPCPublisher...
[WARN] - Could not stop module OPCPublisher
[WARN] -         caused by: Target of operation already in this state
[INFO] - [mgmt] - - - [2021-04-01 18:17:37.379090020 UTC] "POST /modules/OPCPublisher/stop?api-version=2020-07-07 HTTP/1.1" 304 Not Modified 0 "-" "-" auth_id(-)
[INFO] - Updating module OPCPublisher
[INFO] - Removing module OPCPublisher...
[INFO] - Successfully removed module OPCPublisher
[INFO] - Pulling image mcr.microsoft.com/iotedge/opc-publisher:2.7.206...
[INFO] - Successfully pulled image mcr.microsoft.com/iotedge/opc-publisher:2.7.206
[INFO] - Creating module OPCPublisher...
[INFO] - Successfully created module OPCPublisher
[INFO] - [mgmt] - - - [2021-04-01 18:17:38.006553117 UTC] "PUT /modules/OPCPublisher?api-version=2020-07-07 HTTP/1.1" 200 OK 1032 "-" "-" auth_id(-)
[INFO] - Starting module OPCPublisher...
[INFO] - Successfully started module OPCPublisher
[INFO] - [mgmt] - - - [2021-04-01 18:17:39.299426975 UTC] "POST /modules/OPCPublisher/start?api-version=2020-07-07 HTTP/1.1" 204 No Content - "-" "-" auth_id(-)
[INFO] - [work] - - - [2021-04-01 18:17:39.375957395 UTC] "POST /modules/%24edgeAgent/genid/637515229146279910/encrypt?api-version=2020-07-07 HTTP/1…"-" "-" auth_id(-)
[INFO] - [mgmt] - - - [2021-04-01 18:17:44.522313202 UTC] "GET /modules?api-version=2020-07-07 HTTP/1.1" 200 OK 1647 "-" "-" auth_id(-)
[INFO] - [mgmt] - - - [2021-04-01 18:17:49.676686278 UTC] "GET /modules?api-version=2020-07-07 HTTP/1.1" 200 OK 1647 "-" "-" auth_id(-)
[INFO] - Stopping module OPCPublisher...
[WARN] - Could not stop module OPCPublisher
[WARN] -         caused by: Target of operation already in this state
[INFO] - [mgmt] - - - [2021-04-01 18:17:49.686432844 UTC] "POST /modules/OPCPublisher/stop?api-version=2020-07-07 HTTP/1.1" 304 Not Modified 0 "-" "-" auth_id(-)
[INFO] - Starting module OPCPublisher...
[INFO] - Successfully started module OPCPublisher
[INFO] - [mgmt] - - - [2021-04-01 18:17:50.994808203 UTC] "POST /modules/OPCPublisher/start?api-version=2020-07-07 HTTP/1.1" 204 No Content - "-" "-" auth_id(-)
[INFO] - [work] - - - [2021-04-01 18:17:51.071914677 UTC] "POST /modules/%24edgeAgent/genid/637515229146279910/encrypt?api-version=2020-07-07 HTTP/1…"-" "-" auth_id(-)
[INFO] - [mgmt] - - - [2021-04-01 18:17:56.089069332 UTC] "GET /modules?api-version=2020-07-07 HTTP/1.1" 200 OK 1647 "-" "-" auth_id(-)
[INFO] - [mgmt] - - - [2021-04-01 18:18:01.239688710 UTC] "GET /modules?api-version=2020-07-07 HTTP/1.1" 200 OK 1647 "-" "-" auth_id(-)
[INFO] - Checking edge runtime status
[INFO] - Edge runtime is running.
```

koepalex, 2021-04-07:
Currently the ARM32v7 of OPC Publisher are not available on Microsoft Container Registry. This is most likely the issue you are facing. We are working on this issue.

marcschier, 2021-04-08:
#1085

koepalex, 2021-04-08:
Currently ARM64 isn't supported but we will handle this as an feature request.

closed 2023-01-16 by marcschier