Azure/Industrial-IoT #1090: AzureDevops Deploy Intergration

Mikael159, 2021-04-07:
Hello,

I would like to integrate the Azure Industrial iot deployment into a devops chain.
We are currently using AzureDevops for our Deployment Pipeline.

How do I add the deployment to my devops chain?

I tried with the statement "deploy.cmd -type minimum -version 2.7.182 [...]" but it got stuck at the "Signing in ..." step.

Do you have a way to do it?

thank you, 
best regards

Mikael159, 2021-04-07:
step to reproduce :

- checkout: git://# Azure Repos Git repository / clone of AzureIndustrialIot
  displayName: 'AIIoT Checkout Industrial-IoT'
- task: AzurePowerShell@4
  displayName: 'AIIoT Azure CLI Script Test'
  inputs:
    azureSubscription: ${{ parameters.TargetSubscriptionConnection }}
    scriptType: inlineScript
    azurePowerShellVersion: 5.3.0
    inline: |
     $(Agent.BuildDirectory)\s\deploy.cmd -type minimum -version 2.7.182  [...]

marcschier, 2021-04-09:
Requires service principal authentication, which currently is not supported by the script.  This is on our backlog.  Please take a look at the end to end test automation and how it deploys from a pipeline as a mitigation for the time being.

marcschier, 2023-05-12:
This will be possible to some degree in 2.9 release, see \Industrial-IoT\tools\e2etesting\steps\deployplatform.yaml.  You must change this slightly to precreate the applications in AAD and grant admin consent, unless you use a testing only AAD tenant and give the SP the permissions.

closed 2023-05-19 by marcschier