Azure/Industrial-IoT #1093: Swagger microservice uses 10 GB of file storage in App Service plan

ajithkumars, 2021-04-09:
To categorize correctly as you feel.
The issue :  swagger microservice container has used 10 GB of app plan storage (of 10 GB possible), which is shared along with engineering tool (Azure IIoT Platform 2.7.183). 
Context: Industrial IOT 2.7.183 deployed using scripts for full deployment. Has been working fine for 1 and half months.
We use both engineering tool and swagger for our purpose.
Additional question: What is using up this amount of space in swagger service, and how can we clean this up to continue using the service?

**To Reproduce**
We have been running this now for around 2 months (so that is what leads to it or probably other issues in between), so uncertain on this.

**Expected behavior**
We do expect that swagger microservice uses as much storage as engineering tool which is around 14 MB or atleast not in the scale of GB.

[image]

hansgschossmann, 2021-04-13:
@ajithkumars can you please try to check how the file system container is using the 10GB? it would be of interest which directories are holding majority of the data. this information will help us to properly identify the root cause.

ajithkumars, 2021-04-13:
Have researched further on this issue, found a possibility to login to the container and check where it is used.
This is possible at:
https://{ApplicationName}.scm.azurewebsites.net/DebugConsole

How to get here from service yourself:
{ApplicationName} -> Development Tools -> Advanced Tools -> Go -> Bash

[image]

Here are the logs under / home /LogFiles. (How to restrict logging by space used here?)
Exception that is causing this issue which repeats every second : (Stack Trace)

2021-04-13T02:52:31.356337384Z [02:52:31 WRN Microsoft.Azure.IIoT.Hub.Processor.Services.EventProcessorFactory] EventProcessor 5030dc08-6e3e-4594-b25f-6a89659fa95e for partition 0 error 
2021-04-13T02:52:31.357022534Z Microsoft.Azure.EventHubs.MessagingEntityNotFoundException: The messaging entity 'iothub-ns-iothub-wu4-8611053-45a13a5daf:eventhub:iothub-wu4lqe~8191|tunnel' could not be found. To know more visit https://aka.ms/sbResourceMgrExceptions.  TrackingId:e0de1f54-6e48-4c27-a4e4-a558db94a3f0_B1, SystemTracker:iothub-ns-iothub-wu4-8611053-45a13a5daf:eventhub:iothub-wu4lqe~8191|tunnel, Timestamp:2021-04-13T02:52:31 TrackingId:858c7696cf4d4da1ae168cbd4b37e596_G0, SystemTracker:gateway5, Timestamp:2021-04-13T02:52:31
 at Microsoft.Azure.EventHubs.Amqp.AmqpPartitionReceiver.OnReceiveAsync(Int32 maxMessageCount, TimeSpan waitTime)
 at Microsoft.Azure.EventHubs.Amqp.AmqpPartitionReceiver.OnReceiveAsync(Int32 maxMessageCount, TimeSpan waitTime)
 at Microsoft.Azure.EventHubs.PartitionReceiver.ReceiveAsync(Int32 maxMessageCount, TimeSpan waitTime)
 at Microsoft.Azure.EventHubs.Amqp.AmqpPartitionReceiver.ReceivePumpAsync(CancellationToken cancellationToken, Boolean invokeWhenNoEvents)

We have seen this before but not sure what is the issue. (The swagger application does work fine for things we have used it despite this issue).

hansgschossmann, 2021-04-20:
@ajithkumars thanks for providing details. we will look into this and come back to you.

hansgschossmann, 2021-12-15:
@ajithkumars we looked at this and it is most likely because of a configuration of of AppService. We will need to analyze that in more detail in the next step.