Azure/Industrial-IoT #1108: Manifest for mcr.microsoft.com/iot/industrial-iot-frontend:preview not found

pablolaurin, 2021-04-17:
**Describe the bug**

I can't pull "mcr.microsoft.com/iot/industrial-iot-frontend:preview"

**To Reproduce**

1. Go to command prompt
2. Execute  'docker pull mcr.microsoft.com/iot/industrial-iot-frontend:preview'
3. See error:

"preview: Pulling from iot/industrial-iot-frontend
manifest for mcr.microsoft.com/iot/industrial-iot-frontend:preview not found: manifest unknown: manifest sha256:2797cffde6bee3cb801ca11df410729ddd879b46ce8cec4700278b592e612985 is not found" 

**Expected behavior**
The image should be pulled without errors.

**Desktop:**
- OS: Windows 10 Enterprise Version 1909 (OS Build 18363.1441)
- Docker Engine: v20.10.5

**Additional context**
The error also appears when trying to deploy Industrial-Iot in the cloud using ".\deploy" as described at https://github.com/Azure/Industrial-IoT#readme

pablolaurin, 2021-04-17:
However, image tagged "latest" works:

`docker pull mcr.microsoft.com/iot/industrial-iot-frontend:latest`

Result:

```
latest: Pulling from iot/industrial-iot-frontend
45b42c59be33: Already exists                                                                                            fcffb47bf4a3: Already exists                                                                                            3e05eab0cc9a: Pull complete                                                                                             1c41c24e4de6: Pull complete                                                                                             cb7026bb9bf2: Pull complete                                                                                             57c39575d2af: Pull complete                                                                                             1fd7cce923c2: Pull complete                                                                                             Digest: sha256:1ed8879d1de6971b4458a96713cf86b0c6f730cfa52e78f04c300f4e3e8e424c
Status: Downloaded newer image for mcr.microsoft.com/iot/industrial-iot-frontend:latest
mcr.microsoft.com/iot/industrial-iot-frontend:latest
```
What's the difference between "latest" and "preview". Can I deploy "latest"?

marcschier, 2021-05-19:
Main branch now deploys latest.

closed 2021-05-19 by marcschier