Azure/Industrial-IoT #1132: Readme Misinformation

suneetnangia, 2021-04-29:
**Describe the bug**
Readme mentions "and run on managed Azure Kubernetes Services (currently in preview)", this could be misleading as AKS is in GA.

hansgschossmann, 2021-05-04:
We see on this page (https://azure.github.io/Industrial-IoT/deploy/):
Azure Kubernetes Services should be used to host the cloud micro-services (this is currently in preview, please see our documentation).

@suneetnangia is this the location you are referring to? If not, pls post a link to the page you saw this.

suneetnangia, 2021-05-05:
Hi Hans, I saw it here https://github.com/Azure/Industrial-IoT, I am not sure why we are suggesting it's in preview as AKS is not in preview. If a correction is needed, we should also fix it at the location you specified as well.

hansgschossmann, 2021-05-11:
@suneetnangia this is a doc bug. we brought it to the attention of the team.

marcschier, 2021-05-19:
Fixed in linked commit.

closed 2021-05-19 by marcschier