Azure/Industrial-IoT #1133: Heartbeat not working in newer versions of OPC Publisher

ajithkumars, 2021-04-29:
We are currently using full Industrial IOT deployment version 2.7.183. And this bug/issue concerns opc publisher.
Heartbeat does not work as we expect it to work, i.e, static values are not read in every hearbeat interval.
We tested different versions to be sure:
2.5: Works
2.6: Works
2.7: Does not work
2.7.183: Does not work
2.7.206: Does not work (latest version)

Steps to reproduce the behavior:
1. Deploy Industrial IOT (2.7.183)
2. Add an edge device.
3. Add opc publisher with heartbeat interval set (also has a opc server it can read from)
4. Depoy different versions of the pulisher itself to check it works differently 

**Expected behavior**
When heartbeat interval for example 10 seconds is set, publisher should retrieve even static data (data that does not change)
every 10 seconds and send to IoTHub.

Edge running on 
 - OS: Ubuntu 20.04 Desktop (Running on VM)

koepalex, 2021-08-19:
Could you please share some more details? I've just tested it with iot-edge-opc-plc with 60 nodes and it worked.

ajithkumars, 2021-08-19:
Which version did you test ?

koepalex, 2021-08-20:
2.8.0 (LTS)

ajithkumars, 2021-08-20:
Right, as mentioned in the issue all those versions tested were older. So, its great that it is fixed in the new version (if it is)

koepalex, 2021-08-20:
Just for my understanding. I just tested standalone (because you mentioned 2.5 and this is only available for standalone). Your testing of 2.6/2.7 was in standalone or orchestrated mode (as part of Industrial IoT Platform) ?

ajithkumars, 2021-08-20:
It has been a while now, but I think as described it is part of the orchestrated mode.

koepalex, 2021-08-20:
I've tested also 2.8 in orchestrated mode and it works now.

ajithkumars, 2021-08-20:
Thank you, I will look into it.