Azure/Industrial-IoT #1148: PubSub JSON DataSetWriterId should be a UInt16, not a String

mregen, 2021-05-05:
**Describe the bug**
In this Mantis Issue:

4327 | DataSetWriterId field in JSON   DataSetMessage | Changed DataType of the   DataSetWriterId in a JSON DataSetMessage from String to UInt16 to be   consistent with other definitions.
-- | -- | --

the datatype was changed from string to UInt16.

**To Reproduce**
Send PubSub JSON encoded message to cloud, value is set to string like:
"DataSetWriterId": "uat46f9f8f82fd5c1b42a7de31b5dc2c11ef418a62f",

**Expected behavior**
DatasetWriterId is a UInt16

**Additional context**
Add any other context about the problem here.

hansgschossmann, 2021-09-21:
test

marcschier, 2023-01-17:
In 2.9.0 with --strict, DataSetWriterId is int, DataSetWriterName is string.  Without --strict old behavior applies.

closed 2023-01-17 by marcschier