Azure/Industrial-IoT #1153: Cloud services producing lots of logs

maalliet, 2021-05-06:
Installed the all-in-one solution of the Industrial IoT Cloud services.
The API services seem to produce a lot of logs - majority of them are informational, therefore costs increasing quite significantly.
Is there a way to chance the loglevel to warning without needing to rebuild the solution?

maalliet, 2021-05-18:
Hi @hansgschossmann 

From my point of view it is partially related. the LogFiles folder is on my side also increasing in size over time. However, I do not receive any error message in the logs. 
Underlying services are producing similar informational logs over and over again (see small sample below). Would be great if the loglevel of the application could be configured to warning or above. This will decrease the amount of logs significantly, although it does not resolve the increasing LogFiles directory size in the long term.

```
2021-05-17T21:39:57.458782466Z [21:39:57 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Updating orchestrator urls... 
2021-05-17T21:39:57.657452604Z [21:39:57 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Job orchestrator url updated to https://APISERVICENAME.azurewebsites.net/edge/publisher on all twins. 
2021-05-17T21:39:57.658315204Z [21:39:57 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Orchestrator Url update finished. 

2021-05-17T21:40:05.223900792Z [21:40:05 INF Microsoft.Azure.IIoT.Auth.IoTHub.TwinIdentityTokenUpdater] Updating identity tokens... 
2021-05-17T21:40:05.660330633Z [21:40:05 INF Microsoft.Azure.IIoT.Hub.Client.IoTHubServiceHttpClient] IOTEDGENAME (publisher) created or updated (AAAAAAAAVCY= -> AAAAAAAAVCc=) 
2021-05-17T21:40:05.779117681Z [21:40:05 INF Microsoft.Azure.IIoT.Auth.IoTHub.TwinIdentityTokenUpdater] Identity Token update finished. 

2021-05-17T21:40:57.659263260Z [21:40:57 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Updating orchestrator urls... 
2021-05-17T21:40:57.803974419Z [21:40:57 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Job orchestrator url updated to https://APISERVICENAME.azurewebsites.net/edge/publisher on all twins. 
2021-05-17T21:40:57.803999919Z [21:40:57 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Orchestrator Url update finished. 

2021-05-17T21:41:14.686683039Z [21:41:14 INF Microsoft.Azure.IIoT.OpcUa.Registry.Services.ActivationSyncHost] Running endpoint synchronization... 
2021-05-17T21:41:14.829527690Z [21:41:14 INF Microsoft.Azure.IIoT.OpcUa.Registry.Services.ActivationSyncHost] Endpoint synchronization finished. 

2021-05-17T21:41:57.804141670Z [21:41:57 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Updating orchestrator urls... 
2021-05-17T21:41:57.933742856Z [21:41:57 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Job orchestrator url updated to https://APISERVICENAME.azurewebsites.net/edge/publisher on all twins. 
2021-05-17T21:41:57.934626958Z [21:41:57 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Orchestrator Url update finished. 

2021-05-17T21:42:57.934396746Z [21:42:57 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Updating orchestrator urls... 
2021-05-17T21:42:58.084869492Z [21:42:58 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Job orchestrator url updated to https://APISERVICENAME.azurewebsites.net/edge/publisher on all twins. 
2021-05-17T21:42:58.084890592Z [21:42:58 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Orchestrator Url update finished. 

2021-05-17T21:43:15.350216543Z [21:43:15 INF Microsoft.Azure.IIoT.OpcUa.Registry.Services.ActivationSyncHost] Running endpoint synchronization... 
2021-05-17T21:43:16.360090538Z [21:43:16 INF Microsoft.Azure.IIoT.OpcUa.Registry.Services.ActivationSyncHost] Endpoint synchronization finished. 

2021-05-17T21:43:58.081969749Z [21:43:58 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Updating orchestrator urls... 
2021-05-17T21:43:58.236196843Z [21:43:58 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Job orchestrator url updated to https://APISERVICENAME.azurewebsites.net/edge/publisher on all twins. 
2021-05-17T21:43:58.237166847Z [21:43:58 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Orchestrator Url update finished. 

2021-05-17T21:44:58.238895360Z [21:44:58 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Updating orchestrator urls... 
2021-05-17T21:44:58.389988628Z [21:44:58 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Job orchestrator url updated to https://APISERVICENAME.azurewebsites.net/edge/publisher on all twins. 
2021-05-17T21:44:58.391164633Z [21:44:58 INF Microsoft.Azure.IIoT.Agent.Framework.Jobs.JobOrchestratorEndpointSync] Orchestrator Url update finished. 

2021-05-17T21:45:05.778959263Z [21:45:05 INF Microsoft.Azure.IIoT.Auth.IoTHub.TwinIdentityTokenUpdater] Updating identity tokens... 
2021-05-17T21:45:06.083922357Z [21:45:06 INF Microsoft.Azure.IIoT.Hub.Client.IoTHubServiceHttpClient] IOTEDGENAME (publisher) created or updated (AAAAAAAAVCg= -> AAAAAAAAVCk=) 
2021-05-17T21:45:06.202943084Z [21:45:06 INF Microsoft.Azure.IIoT.Auth.IoTHub.TwinIdentityTokenUpdater] Identity Token update finished.
```

marcschier, 2023-06-22:
App service container service now uses ILogger and can be configured using appsettings.json.

closed 2023-06-22 by marcschier