Azure/Industrial-IoT #1182: Failed to deploy Industrial IoT solution. The value of deployment parameter 'aiiotImageNamespace' is null.

FBuchholz, 2021-05-23:
**Describe the bug**
By installation / Setup the IoT Platform for production, we use the win64 binary. At step "Helmsettings: Tag=2.7.206" the depolyment stopped with "Failed to deploy Industrial IoT solution. The value of deployment parameter 'aiiotImageNamespace' is null." error

**To Reproduce**
Steps to reproduce the behavior:
1. Download Microsoft.Azure.IIoT.Deployment Binaries from https://github.com/Azure/Industrial-IoT/blob/main/docs/deploy/howto-deploy-aks.md
2. execute Binaries on windows 10 client
3. See error

**Expected behavior**
[14:59:40 ERR] Failed to deploy Industrial IoT solution.
Microsoft.Rest.Azure.CloudException: The value of deployment parameter 'aiiotImageNamespace' is null. Please specify the value or use the parameter reference. See https://aka.ms/resource-manager-parameter-files for details.
   at Microsoft.Azure.Management.ResourceManager.Fluent.DeploymentsOperations.BeginCreateOrUpdateWithHttpMessagesAsync(String resourceGroupName, String deploymentName, DeploymentInner parameters, Dictionary`2 customHeaders, CancellationToken cancellationToken)
   at Microsoft.Azure.Management.ResourceManager.Fluent.DeploymentsOperations.CreateOrUpdateWithHttpMessagesAsync(String resourceGroupName, String deploymentName, DeploymentInner parameters, Dictionary`2 customHeaders, CancellationToken cancellationToken)
   at Microsoft.Azure.Management.ResourceManager.Fluent.DeploymentsOperationsExtensions.CreateOrUpdateAsync(IDeploymentsOperations operations, String resourceGroupName, String deploymentName, DeploymentInner parameters, CancellationToken cancellationToken)
   at Microsoft.Azure.IIoT.Deployment.Infrastructure.ResourceMgmtClient.CreateResourceGroupDeploymentAsync(IResourceGroup resourceGroup, String deploymentName, Object template, Object parameters, DeploymentMode deploymentMode, CancellationToken cancellationToken)
   at Microsoft.Azure.IIoT.Deployment.Infrastructure.ResourceMgmtClient.CreateResourceGroupDeploymentAsync(IResourceGroup resourceGroup, String deploymentName, String templateJson, String parametersJson, DeploymentMode deploymentMode, CancellationToken cancellationToken)
   at Microsoft.Azure.IIoT.Deployment.Infrastructure.ResourceMgmtClient.CreateResourceGroupDeploymentAsync(IResourceGroup resourceGroup, String deploymentName, String templateJson, IDictionary`2 parameters, DeploymentMode deploymentMode, IDictionary`2 tags, CancellationToken cancellationToken)
   at Microsoft.Azure.IIoT.Deployment.Deployment.DeploymentExecutor.CreateAzureResourcesAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.IIoT.Deployment.Deployment.DeploymentExecutor.RunFullDeploymentAsync(CancellationToken cancellationToken)
Do you want to delete registered Applications and the Resource Group ? Please select Y[yes] or N[no]
[14:59:40 INF] Pushed IIoT configuration parameters to Key Vault.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser chrome]
 - Version 90.0.4430.212 64 Bit

FBuchholz, 2021-05-23:
[image]

karok2m, 2021-05-25:
@FBuchholz, are you specifying any command like parameters when starting the binary ?

FBuchholz, 2021-05-25:
@karok2m  I just follow the doc's (sample-run) without any parameter.

Sample Run

Sample run of Full deployment without any configuration parameters is described below. Microsoft.Azure.IIoT.Deployment would interactively prompt for all required parameters that are not provided through configuration. And if no authentication configuration is provided, then authentication flow with user credentials will be used.

    Note: 4th step below describes the error that you might get if you are lacking admin consent and Granting Admin Consent walks you through steps to mitigate that problem. You will still need to run the application once before granting the consent and that first run will fail.

    Run Microsoft.Azure.IIoT.Deployment application from command line:

    ./Microsoft.Azure.IIoT.Deployment

karok2m

FBuchholz, 2021-05-26:
Hi, here more informations with used parameter.

[image]

FBuchholz, 2021-05-26:
[image]

FBuchholz, 2021-05-26:
.\Microsoft.Azure.IIoT.Deployment.exe /RunMode=Full /Auth:TenantId=deleted /SubscriptionId=deleted /ResourceGroup:UseExisting=false /ResourceGroup:Name=deleted /SaveEnvFile=true /ResourceGroup:Region=EuropeWest /Auth:AzureEnvironment=AzureGlobalCloud /ApplicationName=deleted

karok2m, 2021-05-27:
@FBuchholz, thank you for the command.

PR #1189 (2021-05-31): IAI: Added default values for image namespace, container registry username and password.
Changes:
* Updated NuGet packages.
* Added default values for image namespace, container registry username and password.
* Changed default Kubernetes version to `1.18.14` as `1.18.10` is no longer supported.

Fixes #1182
commit d269b04b (2021-05-31): IAI: Added default values for image namespace, container registry username and password. (#1189)

Changes:
* Updated NuGet packages.
* Added default values for image namespace, container registry username and password.
* Changed default Kubernetes version to `1.18.14` as `1.18.10` is no longer supported.

Fixes #1182
closed 2021-05-31 by koepalex