Azure/Industrial-IoT #1183: Deployment Script error -  Release 2.8.0 - "Deployment template validation failed: 'The provided value 'Standard_D2as_v4' for the template parameter 'edgeVmSize' at line '62' and column '30' is not valid.

FBuchholz, 2021-05-23:
**Describe the bug**
The deployment 'azuredeploy' failed with error(s). Showing 1 out of 1 error(s).
Status Message: At least one resource deployment operation failed. Please list deployment operations for details. Please see https://aka.ms/DeployOperations for usage details. (Code: DeploymentFailed)

**To Reproduce**
Steps to reproduce the behavior:
perform deployment script execution

**Expected behavior**
-web' updated with required resource access.
Client and services AAD applications registered...

The following username and password can be used to log into the deployed VMs:
sandboxuser
Rp8.BvCsfdxpgTG

Starting deployment...
18:14:43 - The deployment 'azuredeploy' failed with error(s). Showing 1 out of 1 error(s).
Status Message: At least one resource deployment operation failed. Please list deployment operations for details. Please see https://aka.ms/DeployOperations for usage details. (Code: DeploymentFailed)
 - {
  "error": {
    "code": "InvalidTemplate",
    "message": "Deployment template validation failed: 'The provided value 'Standard_D2as_v4' for the template parameter 'edgeVmSize' at line '62' and column '30' is not valid. The parameter value is not part of the allowed value(s): ',Standard_B2s,Standard_B2ms,Standard_A2,Standard_A3,Standard_A4,Standard_A5,Standard_A6,Standard_A7,Standard_A8,Standard_A9,Standard_A10,Standard_A11,Standard_A2_v2,Standard_A4_v2,Standard_A8_v2,Standard_A2m_v2,Standard_A4m_v2,Standard_A8m_v2,Standard_D2,Standard_D3,Standard_D4,Standard_D11,Standard_D12,Standard_D13,Standard_D14,Standard_D2_v2,Standard_D3_v2,Standard_D4_v2,Standard_D5_v2,Standard_D11_v2,Standard_D12_v2,Standard_D13_v2,Standard_D14_v2,Standard_D15_v2,Standard_G1,Standard_G2,Standard_G3,Standard_G4,Standard_G5,Standard_F2,Standard_F4,Standard_F8,Standard_F16'.'.",
    "additionalInfo": [
      {
        "type": "TemplateViolation",
        "info": {
          "lineNumber": 62,
          "linePosition": 30,
          "path": "parameters.edgeVmSize.allowedValues"
        }
      }
    ]
  }
} (Code:BadRequest)
 - {
  "error": {
    "code": "InvalidTemplate",
    "message": "Deployment template validation failed: 'The provided value 'Standard_D2as_v4' for the template parameter 'edgeVmSize' at line '62' and column '30' is not valid. The parameter value is not part of the allowed value(s): ',Standard_B2s,Standard_B2ms,Standard_A2,Standard_A3,Standard_A4,Standard_A5,Standard_A6,Standard_A7,Standard_A8,Standard_A9,Standard_A10,Standard_A11,Standard_A2_v2,Standard_A4_v2,Standard_A8_v2,Standard_A2m_v2,Standard_A4m_v2,Standard_A8m_v2,Standard_D2,Standard_D3,Standard_D4,Standard_D11,Standard_D12,Standard_D13,Standard_D14,Standard_D2_v2,Standard_D3_v2,Standard_D4_v2,Standard_D5_v2,Standard_D11_v2,Standard_D12_v2,Standard_D13_v2,Standard_D14_v2,Standard_D15_v2,Standard_G1,Standard_G2,Standard_G3,Standard_G4,Standard_G5,Standard_F2,Standard_F4,Standard_F8,Standard_F16'.'.",
    "additionalInfo": [
      {
        "type": "TemplateViolation",
        "info": {
          "lineNumber": 62,
          "linePosition": 30,
          "path": "parameters.edgeVmSize.allowedValues"
        }
      }
    ]
  }
} (Code:BadRequest)

CorrelationId: 9c7c05fd-348d-44bf-89aa-02833759935d
Deployment failed.
Try again? [y/n]:
Delete resource group? [y/n]: y

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Win 10

**Additional context**
[image]

Add any other context about the problem here.

hansgschossmann, 2021-05-25:
Please ensure your subscription/region combination does support the VM size. If you have validated it, pls update this issue if it still exist.

marcschier, 2021-05-27:
We need to update the template to either add new VM sizes or likely better make the size a text field.

hansgschossmann, 2021-06-22:
@FBuchholz  can you validate the fix @marcschier has merged. Pls let us know your result.

hansgschossmann, 2021-09-15:
Fix is released with 2.8.0.

closed 2021-09-15 by hansgschossmann