Azure/Industrial-IoT #1190: Helm 0.3.2 deployment document is incomplete

denizetkar, 2021-05-27:
**Describe the bug**
This is a bug with the documents on how to deploy Azure IIoT into an existing Kubernetes cluster using Helm package manager.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the current latest stable Helm deployment documentation [page](https://github.com/Azure/Industrial-IoT/blob/helm_0.3.2/deploy/helm/azure-industrial-iot/README.md#azure-aad-app-registration),
2. Click [create AAD App Registrations](https://github.com/Azure/Industrial-IoT/blob/helm_0.3.2/docs/deploy/howto-register-aad-applications.md) link on that webpage,
3. `ServicesApp` and `ClientsApp` app registrations in Azure Active Directory are omitted with a placeholder text "TBD".

**Expected behavior**
I would expect to see detailed instructions on how to register these 2 applications (ServicesApp, ClientsApp) as recommended by the Helm 0.3.2 deployment documentation [page](https://github.com/Azure/Industrial-IoT/blob/helm_0.3.2/deploy/helm/azure-industrial-iot/README.md#azure-aad-app-registration).

**Screenshots**
There is no need for a screenshot. You can simply visit the links I provided.

**Desktop (please complete the following information):**
Irrelevant.

**Smartphone (please complete the following information):**
Irrelevant.

**Additional context**
No further context needed.

karok2m, 2021-05-31:
@denizetkar, please check the [documentation](https://github.com/Azure/Industrial-IoT/blob/main/docs/deploy/howto-register-aad-applications.md) in the `main` branch. The link will be fixed in the upcoming release.

karok2m, 2021-09-03:
@denizetkar, we have release [`0.4.0`](https://github.com/Azure/Industrial-IoT/releases/tag/helm_0.4.0) version of `azure-industrial-iot` chart. You can find the doc for creating AAD App Registration of that version [here](https://github.com/Azure/Industrial-IoT/blob/helm_0.4.0/docs/deploy/howto-register-aad-applications.md). For the latest version of the doc please refer to `main` branch.

closed 2021-09-03 by karok2m