Azure/Industrial-IoT #1193: Open source license violation?

jkoplo, 2021-05-28:
This project is under an MIT license, but has dependencies on [OPCFoundation/UA-.NETStandard](https://github.com/OPCFoundation/UA-.NETStandard). The license for that project is a dual GPL2.0/RCL where the RCL is only applicable for paying members.

Doesn't that mean that anyone who uses this library (and isn't a member of the OPC Foundation Corporate Members) is violating the GPL?

koepalex, 2021-05-31:
@jkoplo the dual licensing is only applicable for modification to the Core of UA-.NETStandard stack. This is not done by our product, we just use the provided NuGet packages from OPC Foundation that have a different license [see here](https://www.nuget.org/packages/OPCFoundation.NetStandard.Opc.Ua/1.4.366.38/License).
The license for the NuGet packages allow the usage in open source and commercial products, without the requirement of being OPC Foundation member.

jkoplo, 2021-06-01:
Interesting. So the OPC Foundation's libraries are under three licenses. GPL/RCL for source, but a special (permissive) 'Redistributables' license for the built libraries? 

This means that I can use both the OPC Foundation libraries and the Azure/Industrial-IoT libraries/code in a MIT or Apache licensed program or even an internal-only (LOB) application or commercial application?

I appreciate the reply, the licensing is a bit complex here.

koepalex, 2021-06-01:
I'm not sure if Apache is compatible to MIT, but you can use OPCF libraries and Industrial IoT platform in your own MIT project, as well as in internal application as well as commercial license. As far as I understand, but as a warning I'm not a lawyer.

koepalex, 2021-06-01:
@mregen your comment?

jkoplo, 2021-06-01:
Awesome. I'm one of the maintainers of [libplctag.NET](https://github.com/libplctag/libplctag.NET). 

I've long been thinking of smashing that together with an OPC-UA server implementation as a sort of open OPC-UA server. I also have some professional need for OPC-UA serving from C#. 

Seems like all of this should be fair game with the OPCF libraries. Thanks a ton for your insight.

koepalex, 2021-06-02:
libplctag seems to be a cool project. If you want to integrate an OPC UA Server, you could start with [OPC Foundation UA-.NETStandard Samples](https://github.com/OPCFoundation/UA-.NETStandard-Samples).

I will close this issue for now, please reopen if you have further questions.

closed 2021-06-02 by koepalex