Azure/Industrial-IoT #1194: Which development environment is needed for developing "Microsoft.Azure.IIoT.Deployment Binaries"

jzsnd, 2021-05-31:
Hi Experts, 

We would like to develop a software based on this:

https://github.com/Azure/Industrial-IoT/blob/main/docs/deploy/howto-deploy-aks.md
https://github.com/Azure/Industrial-IoT/tree/main/deploy/src/Microsoft.Azure.IIoT.Deployment

Before we start, we would like to know does anyone has done this and could offer us some advice. For example, which development environment and which development tools (for instance, Visio studio? which version, which library) do we need. 

We are looking forward to hearing from you and thanks a lot.
Jiannan

koepalex, 2021-05-31:
Hi Jiannan,

when using the Industrial IoT Platform, there is an Open API (aka swagger) definition of the provided endpoints: [Open API](https://github.com/Azure/Industrial-IoT/tree/main/api/swagger).

This you can use with any compatible tool, e.g. testing with Postman. I would recommend to generate your client code based on this definition by using [AutoRest](https://github.com/Azure/AutoRest). So you are free in the choice of language, tooling you want to use.

Why did you reference the Microsoft.Azure.IIoT.Deployment? It is _just_ and tool to deploy the Industrial IoT Platform into the customer subscription.

koepalex, 2021-06-07:
@jzsnd I hope my answer has help you, I will close the issue. If you have further questions please reopen.

closed 2021-06-07 by koepalex