Azure/Industrial-IoT #1203: Deployment failed when running script - Simulation devices

NilssonCJ, 2021-06-04:
**Describe the bug**
Deployment failed when running script - see **Additional context** for error message.
Happens when I run the regular script.sh and it tries to deploy the simulation devices.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Industrial-IoT folder
2. Run script.sh
3. See error

**Expected behavior**
That it would finish the deployment but I get errors when I deploy instead.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS

**Additional context**
8:35:10 AM - The deployment 'azuredeploy' failed with error(s). Showing 3 out of 7 error(s).
Status Message: At least one resource deployment operation failed. Please list deployment operations for details. Please see https://aka.ms/DeployOperations for usage details. (Code: DeploymentFailed)
 - {
  "error": {
    "code": "BadRequest",
    "message": "The selected VM size 'Standard_DC1s_v2' cannot boot Hypervisor Generation '1'. If this was a Create operation please check that the Hypervisor Generation of the Image matches the Hypervisor Generation of the selected VM Size. If this was an Update operation please select a Hypervisor Generation '1' VM Size."
  }
} (Code:BadRequest)

Status Message: The selected VM size 'Standard_DC1s_v2' cannot boot Hypervisor Generation '1'. If this was a Create operation please check that the Hypervisor Generation of the Image matches the Hypervisor Generation of the selected VM Size. If this was an Update operation please select a Hypervisor Generation '1' VM Size. (Code:BadRequest)

Status Message: Template output evaluation skipped: at least one resource deployment operation failed. Please list deployment operations for details. Please see https://aka.ms/DeployOperations for usage details. (Code:DeploymentOperationFailed)

marcschier, 2021-06-04:
The deploy script does not use the hyper v gen as criteria for selecting the VM sku.  This filter needs to be added to the script.  Also, we might want a script param to allow overriding the automated selection of sku.

marcschier, 2021-09-09:
Since there is no API to query for the VM SKU supporting Gen1, the only quick fix I see is to change the script to hard code a list of VM skus that fall into this category (based on the documentation) and filter those from the candidate list before picking the top one.  The problem is that this might still break again later when the list is updated, but there is no support for Gen2 VM's for the edge supported OS versions.   If someone would like to contribute a fix to the deploy.ps1 script to filter out a static "list of unsupported sku sizes" (and/or alternatively to allow passing the size from the script args), we would be happy to take a contribution, otherwise it will be treated as medium priority (as it should not happen on most subscriptions) for a 2.8.1 or 2.8.2 release depending on schedule.

hansgschossmann, 2021-09-15:
@NilssonCJ based on the last comment from @marcschier i am going to close this one. If you are interested to provide fix, pls reactivate.

closed 2021-09-15 by hansgschossmann