Azure/Industrial-IoT #1205: IAI Kubernetes version

koepalex, 2021-06-08:
**Is your feature request related to a problem? Please describe.**
In the current implementation IAI has an hardcoded version string for the AKS deployment `KUBERNETES_VERSION` in _AksMgmtClient.cs_.  The current version of main branch **1.18.17** is the oldest supported by region europe north, the current version in release/2.7_rc is so old that you can't use IAI from their.

**Describe the solution you'd like**
* a seperate branch for IAI development, there should be only one current version
* (semi)-automatical updates to IAI or (prefered) just using the latest AKS version

**Describe alternatives you've considered**
None so far, any ideas? @karok2m 

**Additional context**
IAI is important tool because it is used for deployment in internal E2E testing.

koepalex, 2021-06-09:
Thanks for the context, as a user yesterday it was strange for me that the oldest possible version of kubernetes was deployed. If I don't specify anything I would assume to get latest.

karok2m, 2021-06-17:
Just received this email

```
We're retiring AKS version 1.18.x on 31 July 2021. We've detected that one or more AKS clusters in your subscription(s) are using Kubernetes v1.18 or lower. We'll continue to support clusters running AKS v1.18x until 30 July 2021, after which AKS v1.18.x will be removed. 
```

karok2m, 2021-06-17:
@koepalex, I also want to mention that automatically selecting the latest minor (as major is `1`) version is not the best idea in my opinion. Kubernetes does deprecate some API versions. So the deployment manifests that have been validated for a specific version of Kubernetes might not work out of the box for some other version. Here you can find list of [Deprecated APIs](https://kubernetes.io/docs/reference/using-api/deprecation-guide/). That is why I believe that there should be a default minor version that IAI uses and not a logic to automatically select the latest one. Though, we can have a parameter which would define whether IAI tries to use the most latest version of Kubernetes or maybe a latest patch version of tested minor version.

closed 2022-11-24 by marcschier