Azure/Industrial-IoT #1222: Update SourceTimestamp using Heartbeat

mwiot, 2021-06-10:
Hi,

the heartbeat function is used to publish static values. But the sourcetimestamp is not updated (re-read) at every heartbeat interval. Is this implemented, or will this be implemented in the publisher? The SourceTimestamp is only updated when publisher is restarted. 

And have you looked into about the possibility of using a DataChangeFilter with DataChangeTrigger in the UA client?

DataChangeTrigger: 

"STATUS_VALUE_1 Report a notification if either the StatusCode or the value change. The Deadband filter can be used in addition for filtering value changes.
                  For floating point values a Server shall check for NaN and only report a single notification with NaN when the value enters the NaN state.
                  This is the default setting if no filter is set.
   **STATUS_VALUE_TIMESTAMP_2 Report a notification if either StatusCode, value or the SourceTimestamp change.** "

Reference: 
[https://reference.opcfoundation.org/v104/Core/docs/Part4/7.17.2/](https://reference.opcfoundation.org/v104/Core/docs/Part4/7.17.2/)

Best regards

mike-romanowski-hgi, 2023-01-10:
Please give an update on the status of this ticket.  If SourceTimestamp is not updated by OPCPublisher when a node is published due to a HeartbeatInterval being used, what from the OPCPublisher message can be used to generate a timestamp for time series data stores for data values that are constant or rarely change.  I realize we can generate out own timestamp at the time the data is inserted into the time series data store, however, that obviously does not reflect the time of the sampling of the data.  It is mentioned in multiple places in the online OPCPublisher documentation that the SourceTimestamp will be updated when a node is published due to HeartbeatInterval.

For instance:
https://learn.microsoft.com/en-us/azure/industrial-iot/tutorial-publisher-performance-memory-tuning-opc-publisher#:~:text=Starting%20with%20OPC,the%20heartbeat%20triggers.

As well as:
https://azure.github.io/Industrial-IoT/modules/publishednodes_2.8.json?raw=1#:~:text=//%20Specifies%20that%20there%20should%20be%20a%20heartbeat%20generated%20by%20OPC%20Publisher%2C%20this%20means%20that%20after%20the%20given%20interval%20the%20last%20message%20will%20be%20sent%20again%20with%20an%20updated%20SourceTimestamp%20value.%0A%20%20%20%20%20%20%20%20//%20Value%20expressed%20in%20seconds.%0A%20%20%20%20%20%20%20%20%22HeartbeatInterval%22%3A%203600%2C

We are running OPCPublisher 2.8.4 and, while the node is being published at the configured HeartbeatInterval, neither the SourceTimestamp or ServerTimestamp are changing unless the data value changes.

Thank you

mwiot, 2023-01-13:
You can close this. You are correct that the sourcetimestamp should be the timestamp from the source in the ua server. 
For our use case we are able to use the iot hub enqueued timestamp to get the "current time" along with the source timestamp.

closed 2023-01-13 by koepalex