Azure/Industrial-IoT #1226: "Manifest unknown" errors pulling docker images

rcowsill, 2021-06-12:
**Describe the bug**
Docker Hub has the following pages for images from this repo:
- https://hub.docker.com/_/microsoft-iot-industrial-iot-auth-service
- https://hub.docker.com/_/microsoft-iot-industrial-iot-users-service

The example `docker pull` commands on those pages all fail with a "manifest unknown" error:

**To Reproduce**
Steps to reproduce the behavior:
1. Go to (eg) https://hub.docker.com/_/microsoft-iot-industrial-iot-auth-service
2. Scroll down to 'Featured Tags'
3. Copy the example command line to pull one of the tags, eg:
   `docker pull mcr.microsoft.com/iot/industrial-iot-users-service:latest`
4. Run the copied command line
5. Observe "manifest unknown" error

**Expected behavior**
Image pull succeeds, or the Docker Hub page is removed/updated to say the images are unavailable.

**Desktop (please complete the following information):**
 - OS: Windows Server 2016
 - Browser: Firefox
 - Version: 89

hansgschossmann, 2021-06-15:
@rcowsill we have been cleaning up MCR a couple of months ago, to remove images which do not belong to any released version. We might need to review if the federation of some containers with dockerhub is still in place and needs to be cleaned up.

Have you seen issues with the Industrial-IoT platform deployment or functionality because of the not working URIs or are you just curious, why they are not there?

marcschier, 2021-06-18:
@rcwsill, correct, these images should not exist, that behavior is ok.  Having the dockerhub documentation still up is incorrect and will be fixed for the next release, tentatively July.

marcschier, 2021-07-14:
2.8.0 released and MCR yaml updated.

closed 2021-07-14 by marcschier