Azure/Industrial-IoT #1228: Onboarding module fails

denizetkar, 2021-06-14:
**Describe the bug**
The **onboarding** module of the cloud microservices fails to function. When I run the `kubectl get pods -n azure-iiot-ns` command, the output is:
```bash
NAME                                              READY   STATUS    RESTARTS   AGE
azure-iiot-edge-jobs-b98fd4bcb-tb2v8              1/1     Running   0          38s
azure-iiot-events-79848fbcbf-9g8sj                1/1     Running   0          38s
azure-iiot-events-processor-75c985f9b7-2xlc8      0/1     Error     1          38s
azure-iiot-gateway-756967d779-bsfmr               1/1     Running   0          38s
azure-iiot-history-8999f6796-glhzn                1/1     Running   0          38s
azure-iiot-onboarding-5db46cb77f-hhhc7            0/1     Error     1          38s
azure-iiot-publisher-55988b778d-b57qv             1/1     Running   0          38s
azure-iiot-registry-55b98d9f6c-j5pl6              1/1     Running   0          38s
azure-iiot-sync-5c4479cd49-dnxjv                  1/1     Running   0          38s
azure-iiot-telemetry-processor-5647594c59-cc6g6   0/1     Error     1          38s
azure-iiot-tunnel-processor-7954d89fb8-k2hph      0/1     Error     1          38s
azure-iiot-twin-7d5b98fd9d-fkpkf                  1/1     Running   0          38s
```
Also if I take a peek into the logs, `kubectl logs kubectl logs azure-iiot-onboarding-5db46cb77f-hhhc7 -n azure-iiot-ns`:
```bash
[18:09:12 INF Microsoft.Azure.IIoT.Hub.Processor.EventHub.EventProcessorHost] Using Consumer Group: "onboarding" 
[18:09:12 INF Microsoft.Azure.IIoT.AspNetCore.Diagnostics.Default.MetricServerHost] Started prometheus at 9501/metrics 
[18:09:12 INF Microsoft.Azure.IIoT.Messaging.Default.EventBusHost] Event bus host running.
[18:09:13 ERR Microsoft.Azure.IIoT.Hub.Processor.EventHub.EventProcessorHost] Error starting event processor host.
Microsoft.Azure.EventHubs.Processor.EventProcessorConfigurationException: Encountered error while fetching the list of EventHub PartitionIds
 ---> Microsoft.Azure.EventHubs.MessagingEntityNotFoundException: The messaging entity 'sb://ihsuprodamres112dednamespace.servicebus.windows.net/iot-hub-materialfluss' could not be found. To know more visit https://aka.ms/sbResourceMgrExceptions.
   at Microsoft.Azure.EventHubs.Amqp.Management.AmqpServiceClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.EventHubClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()
   --- End of inner exception stack trace ---
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.InitializeStoresAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.StartAsync()
   at Microsoft.Azure.EventHubs.Processor.EventProcessorHost.RegisterEventProcessorFactoryAsync(IEventProcessorFactory factory, EventProcessorOptions processorOptions)
   at Microsoft.Azure.IIoT.Hub.Processor.EventHub.EventProcessorHost.StartAsync() in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Processor\src\EventHub\EventProcessorHost.cs:line 101
[18:09:13 ERR Microsoft.Azure.IIoT.Services.Processor.Onboarding.HostStarterService] Failed to start some hosts.
Microsoft.Azure.EventHubs.Processor.EventProcessorConfigurationException: Encountered error while fetching the list of EventHub PartitionIds
 ---> Microsoft.Azure.EventHubs.MessagingEntityNotFoundException: The messaging entity 'sb://ihsuprodamres112dednamespace.servicebus.windows.net/iot-hub-materialfluss' could not be found. To know more visit https://aka.ms/sbResourceMgrExceptions.
   at Microsoft.Azure.EventHubs.Amqp.Management.AmqpServiceClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.EventHubClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()
   --- End of inner exception stack trace ---
   at Microsoft.Azure.IIoT.Hub.Processor.EventHub.EventProcessorHost.StartAsync() in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Processor\src\EventHub\EventProcessorHost.cs:line 106
   at Microsoft.Azure.IIoT.Services.Processor.Onboarding.HostStarterService.StartAsync(CancellationToken cancellationToken) in D:\a\1\s\services\src\Microsoft.Azure.IIoT.Services.Processor.Onboarding\src\HostStarterService.cs:line 79
Unhandled exception. Microsoft.Azure.EventHubs.Processor.EventProcessorConfigurationException: Encountered error while fetching the list of EventHub PartitionIds
 ---> Microsoft.Azure.EventHubs.MessagingEntityNotFoundException: The messaging entity 'sb://ihsuprodamres112dednamespace.servicebus.windows.net/iot-hub-materialfluss' could not be found. To know more visit https://aka.ms/sbResourceMgrExceptions.
   at Microsoft.Azure.EventHubs.Amqp.Management.AmqpServiceClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.EventHubClient.GetRuntimeInformationAsync()
   at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()
   --- End of inner exception stack trace ---
   at Microsoft.Azure.IIoT.Hub.Processor.EventHub.EventProcessorHost.StartAsync() in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Processor\src\EventHub\EventProcessorHost.cs:line 106
   at Microsoft.Azure.IIoT.Services.Processor.Onboarding.HostStarterService.StartAsync(CancellationToken cancellationToken) in D:\a\1\s\services\src\Microsoft.Azure.IIoT.Services.Processor.Onboarding\src\HostStarterService.cs:line 79
   at Microsoft.Extensions.Hosting.Internal.Host.StartAsync(CancellationToken cancellationToken)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
   at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.Run(IHost host)
   at Microsoft.Azure.IIoT.Services.Processor.Onboarding.Program.Main(String[] args) in D:\a\1\s\services\src\Microsoft.Azure.IIoT.Services.Processor.Onboarding\src\Program.cs:line 47
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install docker desktop 3.4.0 with docker version 20.10.7,
2. Install minikube v1.20.0 and run `minikube start`,
3. Deploy Azure IIoT cloud microservices with helm using [the documentation](https://github.com/Azure/Industrial-IoT/blob/helm_0.3.2/deploy/helm/azure-industrial-iot/README.md),
    * Provision all required Azure services,
    * Perform Azure AAD app registration,
    * Install the helm chart into the minikube cluster.
4. After some minutes, observe using `kubectl` that the onboarding pod is failing with errors.

**Expected behavior**
I would expect the onboarding agent pod to be in "Running" state just as all the other pods, instead of "Error" state.

**Desktop (please complete the following information):**
 - OS: Windows 10 build 2004

**Additional context**
Not sure if more is needed. I will provide more context if asked.

hansgschossmann, 2021-06-15:
@denizetkar which version of the Industrial-IoT platform are you using?

denizetkar, 2021-06-17:
Do you need more information? Because, there is still a `need more information` label attached to this issue. Otherwise, I would like it if you remove the label, please,

denizetkar, 2021-06-20:
I think it might be helpful if I also provide you the `values.yaml` that I gave to `helm` as input for the deployment:

```yaml
deployment:
  ingress:
    annotations:
      nginx.ingress.kubernetes.io/proxy-send-timeout: "3600"
      nginx.ingress.kubernetes.io/session-cookie-max-age: "14400"
      kubernetes.io/ingress.class: nginx
      nginx.ingress.kubernetes.io/session-cookie-name: affinity
      nginx.ingress.kubernetes.io/session-cookie-expires: "14400"
      nginx.ingress.kubernetes.io/affinity: cookie
      nginx.ingress.kubernetes.io/proxy-read-timeout: "3600"
    enabled: true
azure:
  signalR:
    connectionString: <signalr-connection-string>
    serviceMode: Default
  iotHub:
    sharedAccessPolicies:
      iothubowner:
        connectionString: <iothubowner-connection-string>
    eventHub:
      endpoint: <iothub-eventhub-compatible-endpoint>
      consumerGroup:
        onboarding: onboarding
        events: events
        tunnel: tunnel
        telemetry: telemetry
  storageAccount:
    connectionString: <storage-account-connection-string>
  keyVault:
    uri: <keyvault-connection-string>
  tenantId: <tenant-id>
  cosmosDB:
    connectionString: <cosmosdb-connection-string>
  eventHubNamespace:
    sharedAccessPolicies:
      rootManageSharedAccessKey:
        connectionString: <eventhubnamespace-connection-string>
    eventHub:
      name: <eventhub-name>
      consumerGroup:
        telemetryUx: telemetry_ux
        telemetryCdm: telemetry_cdm
  serviceBusNamespace:
    sharedAccessPolicies:
      rootManageSharedAccessKey:
        connectionString: <servicebusnamespace-connection-string>
  auth:
    servicesApp:
      appId: <service-app-id>
      audience: <service-app-audience>
      secret: <service-app-secret>
    clientsApp:
      appId: <client-app-id>
      secret: <client-app-secret>
    required: true
```

Aside from the fields that I left out for privacy reasons like so `<...>`, all the other fields are exactly what I used in my `values.yaml` file.

hansgschossmann, 2021-06-22:
@denizetkar that is sufficient information. minikube is not a supported scenario. we are going to look into this with low priority when we have time. if you have any logs which are indicating an error condition, pls add it as comment.

denizetkar, 2021-06-24:
Tried the same deployment on a 1 node K8s cluster bootstrapped with `kubeadm`. Following are the details:
**Platform**
**Arch:** AMD64
**OS:** Ubuntu 20.04.2 LTS
**Virtual Machine:** yes

**Steps taken to set up K8s cluster:**
1. Installed moby engine with `sudo apt-get install moby-engine`,
2. Now that I have access to `docker` and `containerd` commands and that I have installed a container runtime, I started to execute the official K8s instructions on how to bootstrap [HERE](https://kubernetes.io/docs/setup/production-environment/tools/kubeadm/install-kubeadm/):
    ```bash
    containerd_config_path="./config.toml"
    kubeadm_config_path="./kubeadm-config.yaml"
    
    cat <<EOF | sudo tee /etc/modules-load.d/k8s.conf > /dev/null
    br_netfilter
    EOF
    cat <<EOF | sudo tee /etc/sysctl.d/k8s.conf > /dev/null
    net.bridge.bridge-nf-call-ip6tables = 1
    net.bridge.bridge-nf-call-iptables = 1
    EOF
    sudo sysctl --system
    
    sudo apt-get update
    sudo apt-get install -y apt-transport-https ca-certificates curl
    sudo curl -fsSLo /usr/share/keyrings/kubernetes-archive-keyring.gpg https://packages.cloud.google.com/apt/doc/apt-key.gpg
    echo "deb [signed-by=/usr/share/keyrings/kubernetes-archive-keyring.gpg] https://apt.kubernetes.io/ kubernetes-xenial main" | sudo tee /etc/apt/sources.list.d/kubernetes.list
    sudo apt-get update
    sudo apt-get install -y kubelet kubeadm kubectl
    sudo apt-mark hold kubelet kubeadm kubectl
    
    cat <<EOF | sudo tee /etc/modules-load.d/containerd.conf > /dev/null
    overlay
    br_netfilter
    EOF
    sudo modprobe overlay
    sudo modprobe br_netfilter
    cat <<EOF | sudo tee /etc/sysctl.d/99-kubernetes-cri.conf > /dev/null
    net.bridge.bridge-nf-call-iptables  = 1
    net.ipv4.ip_forward                 = 1
    net.bridge.bridge-nf-call-ip6tables = 1
    EOF
    sudo sysctl --system
    
    sudo mkdir -p /etc/containerd
    # containerd config default | sudo tee /etc/containerd/config.toml
    sudo cp ${containerd_config_path} /etc/containerd/
    sudo systemctl restart containerd
    
    sudo mkdir /etc/docker
    cat <<EOF | sudo tee /etc/docker/daemon.json > /dev/null
    {
      "exec-opts": ["native.cgroupdriver=systemd"],
      "log-driver": "json-file",
      "log-opts": {
        "max-size": "100m",
        "max-file": "3"
      },
      "storage-driver": "overlay2"
    }
    EOF
    sudo systemctl enable docker
    sudo systemctl daemon-reload
    sudo systemctl restart docker
    
    sudo kubeadm init --config ${kubeadm_config_path} --ignore-preflight-errors NumCPU
    mkdir -p $HOME/.kube
    sudo cp /etc/kubernetes/admin.conf $HOME/.kube/config
    sudo chown $(id -un):$(id -gn) $HOME/.kube/config
    
    kubectl taint nodes --all node-role.kubernetes.io/master-
[... 312 lines cut ...]
    [13:59:31 ERR Microsoft.Azure.IIoT.Services.Processor.Onboarding.HostStarterService] Failed to start some hosts. 
    Microsoft.Azure.EventHubs.Processor.EventProcessorConfigurationException: Encountered error while fetching the list of EventHub PartitionIds
     ---> Microsoft.Azure.EventHubs.MessagingEntityNotFoundException: The messaging entity 'sb://ihsuprodamres112dednamespace.servicebus.windows.net/iot-hub-materialfluss' could not be found. To know more visit https://aka.ms/sbResourceMgrExceptions. 
       at Microsoft.Azure.EventHubs.Amqp.Management.AmqpServiceClient.GetRuntimeInformationAsync()
       at Microsoft.Azure.EventHubs.EventHubClient.GetRuntimeInformationAsync()
       at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()
       --- End of inner exception stack trace ---
       at Microsoft.Azure.IIoT.Hub.Processor.EventHub.EventProcessorHost.StartAsync() in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Processor\src\EventHub\EventProcessorHost.cs:line 106
       at Microsoft.Azure.IIoT.Services.Processor.Onboarding.HostStarterService.StartAsync(CancellationToken cancellationToken) in D:\a\1\s\services\src\Microsoft.Azure.IIoT.Services.Processor.Onboarding\src\HostStarterService.cs:line 79
    Unhandled exception. Microsoft.Azure.EventHubs.Processor.EventProcessorConfigurationException: Encountered error while fetching the list of EventHub PartitionIds
     ---> Microsoft.Azure.EventHubs.MessagingEntityNotFoundException: The messaging entity 'sb://ihsuprodamres112dednamespace.servicebus.windows.net/iot-hub-materialfluss' could not be found. To know more visit https://aka.ms/sbResourceMgrExceptions. 
       at Microsoft.Azure.EventHubs.Amqp.Management.AmqpServiceClient.GetRuntimeInformationAsync()
       at Microsoft.Azure.EventHubs.EventHubClient.GetRuntimeInformationAsync()
       at Microsoft.Azure.EventHubs.Processor.PartitionManager.GetPartitionIdsAsync()
       --- End of inner exception stack trace ---
       at Microsoft.Azure.IIoT.Hub.Processor.EventHub.EventProcessorHost.StartAsync() in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Processor\src\EventHub\EventProcessorHost.cs:line 106
       at Microsoft.Azure.IIoT.Services.Processor.Onboarding.HostStarterService.StartAsync(CancellationToken cancellationToken) in D:\a\1\s\services\src\Microsoft.Azure.IIoT.Services.Processor.Onboarding\src\HostStarterService.cs:line 79
       at Microsoft.Extensions.Hosting.Internal.Host.StartAsync(CancellationToken cancellationToken)
       at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.RunAsync(IHost host, CancellationToken token)
[... cut ...]
       at Microsoft.Extensions.Hosting.HostingAbstractionsHostExtensions.Run(IHost host)
       at Microsoft.Azure.IIoT.Services.Processor.Onboarding.Program.Main(String[] args) in D:\a\1\s\services\src\Microsoft.Azure.IIoT.Services.Processor.Onboarding\src\Program.cs:line 47
    ```

This means that the problem is not related to Minikube but a generic problem where IIoT cloud modules cannot be deployed to any private K8s cluster. Do I have to only use Azure Kubernetes for using this software I wonder?

hansgschossmann, 2021-06-29:
@denizetkar thanks for the additional information. We are following up internally and will get back to you.

marcschier, 2021-06-29:
@denizetkar - thanks for providing the detailed information.  Our team has documented how to deploy to [AKS](https://github.com/Azure/Industrial-IoT/blob/main/docs/deploy/howto-deploy-aks.md), i.e., other clusters have not been tested nor are supported (by our team).  

In addition to the documentation you might want to start looking at the code under /deploy folder.  The key is that the micro services require Azure services such as event hub etc. to be deployed to work, and these are deployed using the ARM templates there.  The configuration needs to be added to the values.yaml file.  With the configuration the services will not work as shown in your case. Some documentation related to it can be found [here](https://github.com/Azure/Industrial-IoT/blob/main/docs/deploy/howto-add-aks-to-ps1.md).

closed 2021-06-29 by marcschier

denizetkar, 2021-06-29:
@marcschier Thank you for the response! If you are referring to [this](https://github.com/Azure/Industrial-IoT/issues/1228#issuecomment-864566836) comment of mine, then let me clarify once more that I left the confidential fields out and replaced them with placeholders just to show you how it was structured. It doesn't mean that I didn't deploy or didn't provide the credentials of the running Azure services that are all required by the cloud modules.

As far as I can tell from your comment, you guys don't support deploying IIoT cloud modules on anywhere other than AKS. So, this issue could be closed but I would then kindly ask you to modify your documents to indicate this lack of compatibility and that it is only possible to deploy on AKS.

denizetkar, 2021-06-30:
Just wanted to note my newest findings here for reference. After adding a few more parameters to the `values.yaml` file as below and deploying a TLS secrets component into our local K8s cluster, everything started to work just fine:
```yaml
deployment:
  ingress:
    hostName: <iiot-service-hostname>
    tls:
    - secretName: industrial-iot-tls
      hosts:
      - <iiot-service-hostname>
externalServiceUrl: <iiot-service-hostname>
```
The secrets component look something like:
```yaml
apiVersion: v1
kind: Secret
metadata:
  name: industrial-iot-tls
  namespace: azure-iiot-ns
data:
  tls.crt: <base64-encoded-cert>
  tls.key: <base64-encoded-key>
type: kubernetes.io/tls
```
Now the issue is really resolved and could be actually closed!