Azure/Industrial-IoT #1248: Asset discovery stuck

StevenNoppeVetex, 2021-06-23:
I don't know how I can investigate this issue, but in "asset discovery" after I pressed "Turn on" scanning, the scanning starts but then nothing happens anymore.
A couple of weeks ago this worked fine..

What can I try?

StevenNoppeVetex, 2021-07-05:
> Hello @StevenNoppeVetex can you please provide more information:
> 
> * which version do you have deployed?
2.7.206
> * what is the OS of your IoT Edge device?
Linux Ubuntu
> * which OPC UA servers you were expected to discover and how are they connected the IoT Edge devices?
I have 3 OPC UA servers on the network. One is a Siemens PLC S7-1500, another one is an Ewon Cozy 205, last one is a Windows application that transforms data from machine and convert the data to OPC UA.

For now everything works again.
I was just wondering how I can investigate problems the moment they happen. Log files or something like that?

hansgschossmann, 2021-07-06:
@StevenNoppeVetex it is great to hear that things are working.

When you run into issues like this, then it would be great if you can:
- log into your edge device and collect log files from all modules by running "iotedge support-bundle"
- on the cloud side it depends if you run all microservices in AppService or in aks:
   - for AppService check the logs there
   -  for AKS deployment, you need to check the logs of the onboarding/registry microservices
 
Closing for now, since the issue is no longer there. Pls reactivate if you see it again.

closed 2021-07-06 by hansgschossmann