Azure/Industrial-IoT #1252: OPC certificate management in key vault

emulic, 2021-06-25:
According to [this](https://docs.microsoft.com/en-us/azure/industrial-iot/overview-what-is-industrial-iot#manage-certificates-and-trust-groups), I would expect to be able to manage OPC related certificates (i.e. certificate of the OPC publisher module or certificates of the OPC UA machines) in key vault instead of managing them in the folder structure (which is given by "PkiRootPath=/mount/pki") of the publisher module or mounted folder on edge host. 
However, I'm not able to find any description how to use key vault for OPC certificate management.  
Is it sufficient to just import all required certificates to the key vault (in case of OPC publisher module certificates with the appropriate subject, as described [here](https://github.com/Azure/Industrial-IoT/issues/757#issuecomment-707007047)) or do I need to do something else here?

In general, I'm missing some "get started" documentation regarding certificate management. I.e. I would like to understand what happens with the appropriate OPC Server certificate, when I activate an OPC endpoint by selecting "Turn on" button in the engineering tool? Is it automatically stored to the "trusted" folder? What happens with the certificate in key vault?

marcschier, 2021-06-25:
@emulic - apologies, but the documentation is unfortunately outdated.  The OPC Vault preview has expired, the functionality did not have much uptake, and as a result we decided to remove it (it was in preview).  However the documentation we apparently missed cleaning up.   This means certificates are supported only locally in a volume mount. 

@jehona-m - the documentation should be updated to remove the trust groups/list part, since OPC Vault has been removed (preview/experimental feature).

hansgschossmann, 2022-01-21:
I have removed the reference to OPC Vault i nthe doucmentation: https://github.com/Azure/Industrial-IoT/pull/1527

closed 2022-01-21 by hansgschossmann