Azure/Industrial-IoT #1259: -ms and -si paramters in OPC  Publisher

mahesRBR, 2021-07-01:
Hi,

I posted this question to Azure link 
https://docs.microsoft.com/en-us/answers/questions/450378/-ms-and-si-parameters-in-opc-publisher.html
but was asked to post this message here 
---

I am collecting data from OPC Server via OPC Publisher onto to IoT hub
I need to collect data at 20s frequency and set OpcPublishiNterval as 20000 in PublishedNode.json as shown below

    {
     "Id": "ns=3;i=1008",
     "DisplayName": "F101A",
     "OpcPublishingInterval": 20000
   },

It works fine and I get messages every 20S on IoT hub
But I wish to consolidate the content in each message so the message is sent out by publisher only when it reaches 4Kb limit as standard Azure message is 4Kb and subscription is based on no of messages sent every day

So i set the -ms =4096 while running the publisher . set -si =0 as shown below
{
"Hostname": "publisher",
"Cmd": [
"--pf=/appdata/publishednodes.json",
"--aa",
"--si=0",
"--ms=4096",
"--ct=120"
],
"HostConfig": {
"Binds": [
"/home/ubuntu/Downloads:/appdata"
]
}
}

However get this error for any value set for ms other than 0

[10:20:56 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to create session 'b2c224f8eaa44c2bc65884d1d546a9a378e1771c' due to One or more errors occurred. (BadRequestTimeout) [10:21:06 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Creating session 'b2c224f8eaa44c2bc65884d1d546a9a378e1771c' for endpoint 'opc.tcp://ip-172-31-79-206.ec2.internal:53530/OPCUA/SimulationServer'...

Also tried changing -ct option as error was on create session timeout -but that did not resolve the issue either

Any help is appreciated

Thanks

koepalex, 2021-07-05:
Which version of OPC Publisher are you using? Did you tried `--ms=4096` without `--si` and `--ct` ?

mahesRBR, 2021-07-06:
> [CAUTION: This Email is from outside the Organization. Unless you trust the sender, Don't click links or open attachments as it may be a Phishing email, which can steal your Information and compromise your Computer.]

Hi,
We are using the "latest" version of OPC Publisher . I suppose it does not have a numbered version
The earlier version of OPC Publisher was numbered and it was 2.5.4
thanks

::DISCLAIMER::
The contents of this e-mail and any attachment(s) are confidential and intended for the named recipient(s) only. E-mail transmission is not guaranteed to be secure or error-free as information could be intercepted, corrupted, lost, destroyed, arrive late or incomplete, or may contain viruses in transmission. The e mail and its contents (with or without referred errors) shall therefore not attach any liability on the originator or HCL or its affiliates. Views or opinions, if any, presented in this email are solely those of the author and may not necessarily reflect the views or opinions of HCL or its affiliates. Any form of reproduction, dissemination, copying, disclosure, modification, distribution and / or publication of this message without the prior written consent of authorized representative of HCL is strictly prohibited. If you have received this email in error please delete it and notify the sender immediately. Before opening any email and/or attachments, please check them for viruses and other defects.

hansgschossmann, 2021-07-06:
@mahesRBR Pls use the latest released version of OPC Publisher:  mcr.microsoft.com/iotedge/opc-publisher:2.7.206

The doc for this version is available here: https://github.com/Azure/Industrial-IoT/blob/619f7e867cbd9917a2d450b84eb040155d9b9c29/docs/modules/publisher.md

Please let us know if this and the settings @koepalex has recommended are fixing your problem.

hansgschossmann, 2021-08-05:
@mahesRBR Meanwhile we have been releasing a new version. Please use the tag 2.8.0 for validating if the problem is fixed.

koepalex, 2021-08-16:
@mahesRBR do you have any updates otherwise I think your issues has been solved

mahesRBR, 2021-08-17:
HI,
I have not got a chance to check that
May be I would check in 2 weeks from now
You may close it and we may reopen it again if not working
Thanks for help
regards

koepalex, 2021-08-17:
Please reopen when issue still exist

closed 2021-08-17 by koepalex