Azure/Industrial-IoT #1301: PublishedNodes.Json Schema Validation

suneetnangia, 2021-07-21:
An invalid publishednodes.json file cause LegacyJobOrchestrator in OPC Publisher to hang without surfacing appropriate exceptions, please see the existing issue here https://github.com/Azure/Industrial-IoT/issues/1032 for more details and repro steps. 

We have been working on two prong validation approach for publishednodes.json file for OPC Publisher. 
First one, initial code changes are made in the fork here https://github.com/suneetnangia/Industrial-IoT to validate and throw meaningful exception when schema validation fails. 
Secondly, we have also developed an independent module (not edge module) to generate JSON schema for publishednodes.json programmatically and validate publishednodes.json file outside of publisher itself. This could be useful in multiple ways i.e. where we want to validate publishednodes.json in CICD pipeline as a task, in custom scripts/cli tools or validating UI/auto generated publishednodes.json files.

We'd be keen hear your views on it and next steps to merge it upstream potentially.

CC @WilliamBerryiii @hansgschossmann @koepalex

WilliamBerryiii, 2021-07-26:
I have migrated the tool to GitHub (with backing AzDO build/test) and you can find the repository here: https://github.com/WilliamBerryiii/opcpublisherschemavalidator

Currently, we only support the `publishednodes.json` file, but are actively developing a schema generator for the `telemetryconfiguration.json` file as well.

marcschier, 2021-07-27:
Hi @suneetnangia, @WilliamBerryiii, this is fantastic, we would love to take a contribution and PR.

WilliamBerryiii, 2021-07-27:
Hey @marcschier - @suneetnangia and I will work on preparing a PR and we can discuss the tooling/schema generator in that context and/or in a new issue.  Thanks!

marcschier, 2021-07-29:
Hi @WilliamBerryiii - I looked at the PR, which looks create, but newtonsoft schema library has an incompatible license, which is a show stopper as we cannot redistribute it as part of our containers, nor is it something we want to force customers to pay.  How easy is it to replace with something under MIT or even better coming from MSFT?

WilliamBerryiii, 2021-07-29:
@marcschier - thank you for catching this. We will investigate alternate libraries, ensure their stability and will migrate to a more redistributable-friendly package for schema validation.

hansgschossmann, 2021-09-07:
@WilliamBerryiii have you had a chance to look into the NewtonSoft replacement?

WilliamBerryiii, 2021-09-08:
@hansgschossmann - Yup!  We've implemented JsonSchemaDotNet as a replacement and the [PR is under review](https://github.com/Azure/Industrial-IoT/pull/1305). Thanks again for the support!

hansgschossmann, 2021-09-15:
Thanks. I will follow up to get this reviewed.

PR #1305 (2021-09-22): Add Json Schema Validation for publishednodes.json file
# Description
This PR begins the process of adding JSON-Schema based validation for the Azure OPC Publisher's [`publishednodes.json`](https://raw.githubusercontent.com/Azure/iot-edge-opc-publisher/main/opcpublisher/publishednodes.json) configuration file. A schema generation tool, [the OPC Publisher Schema Validator](https://github.com/WilliamBerryiii/opcpublisherschemavalidator), was created to programmatically generate a json-schema file that adheres to the three (3) supported (documented) formats for OPC UA node configuration within th [... cut]
closed 2021-09-22 by karok2m