Azure/Industrial-IoT #1302: How to run OPC publisher 2.7 on docker

mukeshhope, 2021-07-23:
Hi Team,

I am very new to OPC publisher.
I have setup a OPC simulator server, OPC UA explorer getting live simulation feed.
I want to run OPC Publisher as docker module and connect to OPC UA to get the data.
I found alot of documentation on how to setup from Azure IOT hub but I can not find much documentation on how to run opc publisher as docker container.

As you know I am new, here are some of my questions
1. Do I need Edge runtime, IOT Hub etc to run OPC publisher as a docker container
2. Do I need to use IOTEDGEDEV to create environment before I should try to run docker run mcr.microsoft.com/iotedge/opc-publisher:2.7 <name> as per documentation

Please let me know in detail. I am using windows server 2019 and docker basically windows based container.
I would like to say thankyou in advance

Haeavar, 2021-07-26:
Hi,

we are using the publisher as docker container as well. I start it like that
`docker run -d -v $(pwd)/opcpublisher:/appdata --name opcpublisher {publisherAcr} --pf=/appdata/pn.json --re true --fd=false --ll=Information --mm=PubSub --fm=false --aa --dc="{your device/module connection string from the IoT Hub IoT device}"`

You need to create the pn.json in the "/opcpublisher" directory.
You need to replace the two placeholder parameters with you settings and make changes to the other parameters like "--re" and so on.

Have fun...

mukeshhope, 2021-07-26:
Hi Haeavar,

Thankyou for response, looks like you are running on linux, I needed a windows example

marcschier, 2021-07-27:
Hi, instructions are the same except for the volume mount which needs a folder on windows containing the json file.  

Documentation in the docs/deploy folder is pretty extensive and explains how to deploy Publisher in iot edge, which is the only supported setup right now.  We are however interested to hear what you need, e.g. docker only or kubernetes hosting.  Please open feature request for any missing functionality.

closed 2021-07-27 by marcschier

mukeshhope, 2021-07-28:
Hey Marchchier,

I managed to solve this and you are right
I used below

docker run -v C:\iotedge:c:/appdata mcr.microsoft.com/iotedge/opc-publisher:latest mydevice --dc "mydeviceconnectionstring" --pf=c:\appdata\pn.json --aa
incase someone needs help.