Azure/Industrial-IoT #1304: Could not send worker heartbeat

mukeshhope, 2021-07-26:
Hi Team,

I am trying to run OPC publisher 2.7 in docker from below command
docker run  mcr.microsoft.com/iotedge/opc-publisher:2.7 ht-test --dc <deviceconnectionstring>  -v c:\Mukesh\iotedge:\appdata  --pf=\appdata\pn.json
But I am getting below two errors 

[10:11:51 ERR Root] Unable to start metric server. For more info, please check troubleshooting guide for edge metrics collection
System.Net.HttpListenerException (5): Access is denied.
   at System.Net.HttpListener.AddPrefixCore(String registeredPrefix)
   at System.Net.HttpListener.AddAllPrefixes()
   at System.Net.HttpListener.Start()
   at Prometheus.MetricServer.StartServer(CancellationToken cancel)
   at Prometheus.MetricHandler.Start()
   at Microsoft.Azure.IIoT.Module.Framework.Client.MetricServerEx.StartWhenEnabled(IMetricServer server, IModuleConfig config, ILogger logger) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Hub.Module.Client\src\Extensions\MetricServerEx.cs:line 23
[10:11:52 INF Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory] 0: Device ht-test reconnected due to Connection_Ok.
[10:11:52 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Initialize device twin for ht-test - standalone
[10:11:53 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Applying initial desired state.
[10:11:53 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Reporting currently initial state.
[10:11:53 INF Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost] Module Host started.
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker...
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker...
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker...
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker...
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Creating new worker...
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'ht-test__0'...
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker ht-test__0: {"Type": "publisher", "SiteId": "ht-test", "DeviceId": "ht-test", "ModuleId": null}
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'ht-test__1'...
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker ht-test__1: {"Type": "publisher", "SiteId": "ht-test", "DeviceId": "ht-test", "ModuleId": null}
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'ht-test__2'...
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker ht-test__2: {"Type": "publisher", "SiteId": "ht-test", "DeviceId": "ht-test", "ModuleId": null}
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'ht-test__3'...
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker ht-test__3: {"Type": "publisher", "SiteId": "ht-test", "DeviceId": "ht-test", "ModuleId": null}
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.WorkerSupervisor] Starting worker 'ht-test__4'...
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Worker ht-test__4: {"Type": "publisher", "SiteId": "ht-test", "DeviceId": "ht-test", "ModuleId": null}
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Could not send worker heartbeat.
Microsoft.Azure.IIoT.Exceptions.InvalidConfigurationException: Job orchestrator not configured
   at Microsoft.Azure.IIoT.OpcUa.Api.Publisher.Clients.PublisherOrchestratorClient.SendHeartbeatAsync(HeartbeatModel heartbeat, CancellationToken ct) in D:\a\1\s\api\src\Microsoft.Azure.IIoT.Api\src\Publisher\Clients\PublisherOrchestratorClient.cs:line 84
   at Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker.HeartbeatTimer_ElapsedAsync(Object sender) in D:\a\1\s\common\src\Microsoft.Azure.IIoT.Agent.Framework\src\Agent\Default\Worker.cs:line 151
[10:11:53 INF Microsoft.Azure.IIoT.Agent.Framework.Agent.Worker] Could not send worker heartbeat.
Microsoft.Azure.IIoT.Exceptions.InvalidConfigurationException: Job orchestrator not configured
   at Microsoft.Azure.IIoT.OpcUa.Api.Publisher.Clients.PublisherOrchestratorClient.SendHeartbeatAsync(HeartbeatModel heartbeat, CancellationToken ct) in D:\a\1\s\api\src\Microsoft.Azure.IIoT.Api\src\Publisher\Clients\PublisherOrchestratorClient.cs:line 84

I have created a windows server 2019 with containers machine in Azure and trying to run OPC publisher module here in standalone mode
I have created a IOT HUB Device in Azure
 

 - OS : windows server 2019 with containers

marcschier, 2021-07-27:
Hi @mukeshhope, this happens when the pn json file was not found.  In this case Publisher switches to cloud orchestrated mode and no cloud url was configured.  I noticed that the  volume mapping came after the docker container.  It should come before.  

However, please note that Publisher is only officially supported inside IoT Edge as per documentation.

closed 2021-07-27 by marcschier

mukeshhope, 2021-07-28:
Hi Marcschier,

yes it was happing because of pn.json, it is solved now.
Thankyou for your help